showmewebcam/showmewebcam #2: Webcam settings

PSF1, 2020-09-09:
Hi, thx by this great project : )

I'm using this with a pi zero w and an old 5mp pi cam. 

With ShowMeWebcam the video is dark, when I test de cam with RaspOS I can see the picture ok.

I tried to use a Windows app to set it, but I only can adjust brightness. The default value, 55, is dark, but 80 is all white.
Other settings, like contrast, is not enabled.

[image]

It's possible to enable other settings?

htruong, 2020-09-09:
You can access the camera root prompt:

https://hackaday.io/project/174479-raspberry-pi-0-hq-usb-webcam/log/182789-how-to-tweak-your-camera-settings

After logginf in with root, you can type something like this:

```bash
/usr/bin/v4l2-ctl -c auto_exposure_bias=15
/usr/bin/v4l2-ctl -c contrast=0
```

That should be enough to improve the brightness.

Please report back what values worked for you!

PSF1, 2020-09-09:
Thanks, I will try and then comment :)

PSF1, 2020-09-10:
I get this quality, I think that it's not the best one, but I don't know anything about photography. :_ D

[image]

When I try to set "Codec Controls" I cant with: 
```
# /usr/bin/v4l2-ctl -c h264_level=0
VIDIOC_S_EXT_CTRLS: failed: No such process
h264_level: No such process
```
Is it possible, or necessary, to modify those values?

This is my settings:
```
User Controls

                     brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=50 flags=slider
                       contrast 0x00980901 (int)    : min=-100 max=100 step=1 default=0 value=50 flags=slider
                     saturation 0x00980902 (int)    : min=-100 max=100 step=1 default=0 value=-40 flags=slider
                    red_balance 0x0098090e (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                horizontal_flip 0x00980914 (bool)   : default=0 value=0
                  vertical_flip 0x00980915 (bool)   : default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=3 default=1 value=3
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
                                3: Auto
                      sharpness 0x0098091b (int)    : min=-100 max=100 step=1 default=0 value=100 flags=slider
                  color_effects 0x0098091f (menu)   : min=0 max=15 default=0 value=0
                                0: None
                                1: Black & White
                                2: Sepia
                                3: Negative
                                4: Emboss
                                5: Sketch
                                6: Sky Blue
                                7: Grass Green
                                8: Skin Whiten
                                9: Vivid
                                10: Aqua
                                11: Art Freeze
                                12: Silhouette
                                13: Solarization
                                14: Antique
                                15: Set Cb/Cr
                         rotate 0x00980922 (int)    : min=0 max=360 step=90 default=0 value=0 flags=modify-layout
             color_effects_cbcr 0x0098092a (int)    : min=0 max=65535 step=1 default=32896 value=32896

Codec Controls

             video_bitrate_mode 0x009909ce (menu)   : min=0 max=1 default=0 value=0 flags=update
                                0: Variable Bitrate
                                1: Constant Bitrate
                  video_bitrate 0x009909cf (int)    : min=25000 max=25000000 step=25000 default=10000000 value=25000000
         repeat_sequence_header 0x009909e2 (bool)   : default=0 value=0
            h264_i_frame_period 0x00990a66 (int)    : min=0 max=2147483647 step=1 default=60 value=60
                     h264_level 0x00990a67 (menu)   : min=0 max=11 default=11 value=11
                                0: 1
                                1: 1b
                                2: 1.1
                                3: 1.2
                                4: 1.3
                                5: 2
                                6: 2.1
                                7: 2.2
                                8: 3
                                9: 3.1
                                10: 3.2
                                11: 4
[... 49 lines cut ...]
                                9: Shade
                                10: Greyworld
            image_stabilization 0x009a0916 (bool)   : default=0 value=0
                iso_sensitivity 0x009a0917 (intmenu): min=0 max=4 default=0 value=0
                                0: 0 (0x0)
                                1: 100000 (0x186a0)
                                2: 200000 (0x30d40)
                                3: 400000 (0x61a80)
                                4: 800000 (0xc3500)
           iso_sensitivity_auto 0x009a0918 (menu)   : min=0 max=1 default=1 value=1
                                0: Manual
                                1: Auto
         exposure_metering_mode 0x009a0919 (menu)   : min=0 max=2 default=0 value=2
                                0: Average
                                1: Center Weighted
                                2: Spot
                     scene_mode 0x009a091a (menu)   : min=0 max=13 default=0 value=0
                                0: None
                                8: Night
                                11: Sports

JPEG Compression Controls

            compression_quality 0x009d0903 (int)    : min=1 max=100 step=1 default=30 value=10
```

PSF1, 2020-09-10:
How can I permanent changes? After reboot, it's reset.

htruong, 2020-09-11:
I am going to work on the facility to allow you to save the webcam settings.

kmalinich, 2020-09-14:
I'm using [this camera](https://smile.amazon.com/dp/B00N1YJKFS) which has the original 5MP Raspberry Pi camera sensor, and I had to set the auto exposure bias to 9, but then it looked pretty nice.

tamis-laan, 2020-09-21:
> You can access the camera root prompt:
> 
> https://hackaday.io/project/174479-raspberry-pi-0-hq-usb-webcam/log/182789-how-to-tweak-your-camera-settings
> 
> After logginfg in with root, you can type something like this:
> 
> ```shell
> /usr/bin/v4l2-ctl -c auto_exposure_bias=15
> /usr/bin/v4l2-ctl -c contrast=0
> ```
> 
> That should be enough to improve the brightness. To see all the tweakable parameters, you can list it
> 
> ```shell
> /usr/bin/v4l2-ctl -L | less
> ```
> 
> Please report back what values worked for you!

 Any reason these settings cannot be exposed to the client?

htruong, 2020-09-21:
No, it is just I haven't had time to bake in the ability to set those parameters somewhere.

tamis-laan, 2020-09-21:
I see, well I'm waiting for my HQ camera to arrive, any way I can help you get this done? Any pointers where to look/start?

PSF1, 2020-09-22:
I can help too :_ D
How can we expose the parameters?

htruong, 2020-09-23:
I think the general direction will be to expose a new file with tweakable parameters laid out so the users and freely change and save it. To achieve that:

1. Mount the `/boot` partition on startup (fix `/etc/fstab`).
2. Read a file say `/boot/camera.txt`  and apply each parameter before running the `start-webcam.sh` script.

tamis-laan, 2020-09-25:
This would be to change the default settings? But I assume there should also be some way to change the settings from a menu in the operating system?

kmalinich, 2020-09-26:
@tamis-laan this is a buildroot setup with the wifi intentionally disabled to keep it as lightweight as possible, adding wifi back to it would kind of go against that.. 

@htruong if you used something like `/boot/camera.cfg` as a shell-sourceable file with variables in it (i.e. `CAMERA_CONTRAST=55`, you could source it from `start-webcam.sh` or `multi-gadget.sh`, and in one of those, do something like `/usr/bin/v4l2-ctl -c "contrast=${CAMERA_CONTRAST-50}"` which would use the `${CAMERA_CONTRAST}` value from `/boot/camera.cfg` if it was present, but would use a value of `50` otherwise. Then it would be easily editable by any machine that can read the FAT `/boot` partition.

If you wanted to go overboard you could include some other forms of sanity checks on the values, but without a way to output the result of those checks, it might be a moot point, and therefore the responsibility of the user to not put something like `CAMERA_CONTRAST=1000000` or  `CAMERA_CONTRAST=jdjjdila` in `/boot/camera.cfg` (which IMO is totally fine - if the user can't handle that, they can buy a Logitech USB webcam or something)

htruong, 2020-10-02:
@tamis-laan @kmalinich I can add the wifi + a lightweight server to configure the camera, but that will take some time. I don't think it will make the image much larger, and perhaps we can disable it by default so we don't have to worry about security.

My camera was busted in the last two weeks because the ribbon cable just got bent too much and the camera died while I was using it and I couldn't figure out what was defective. I am working on the issue now as I got another one shipped to me and my setup is working again!

htruong, 2020-10-02:
Would someone help me test this build https://github.com/showmewebcam/showmewebcam/releases/tag/V1.30-preview

tuyenld, 2020-10-02:
@htruong 
I tried to create camera.txt, but it seems the boot partition is Read-Only
```
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                30.9M     30.9M         0 100% /
devtmpfs                193.9M         0    193.9M   0% /dev
tmpfs                   198.2M         0    198.2M   0% /dev/shm
tmpfs                   198.2M      2.8M    195.4M   1% /run
tmpfs                   198.2M         0    198.2M   0% /sys/fs/cgroup
tmpfs                   198.2M      2.8M    195.4M   1% /etc/machine-id
tmpfs                   198.2M         0    198.2M   0% /tmp
tmpfs                    64.0M    152.0K     63.9M   0% /var
/dev/mmcblk0p1           31.9M      9.4M     22.6M  29% /boot

# mount | grep mmcblk0p1
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```

htruong, 2020-10-02:
@tuyenld Sorry I forgot to mention how can you actually do it -- you have two choices:

1. Modify the file in the `boot` partition on your computer.
2. Connect via the ttyACM interface and remount the `boot` partition read-write:

`mount -o remount,rw /boot`

Then you'll be able to edit the file, then reboot.

tuyenld, 2020-10-02:
@htruong 
I tested with Pi Zero W + v1 5M Camera. It worked now!!!
New auto_exposure_bias was applied after reboot.

tamis-laan, 2020-10-02:
I'm will be receiving the last camera components tonight and will give `v1.3` a test run.

PSF1, 2020-10-02:
I created the camera.txt file in /boot and work ok, but if we write a line with spaces around equal fail to read the file or that option. Something like this fail:
`contrast = 0`

but with this work ok:
`contrast=0`

I try adding blank lines and comments with '#' and work ok.

I will write now a file with the default options and comments to put it in the img file if you want it. I forked this repository but I don't know where to put the file. :_ D

PSF1, 2020-10-02:
This is a default file that describe settings, with options and limits.
[default camera.txt](https://github.com/showmewebcam/showmewebcam/files/5319866/default.camera.txt)

Fernyman79, 2020-10-03:
I've tried to test this, but I can't mount the /boot partition. I get the following:

`mount: /boot: mount point does not exist.`

Also, adding a camera.txt by mounting the SD card doesn't work for me. Upon booting the PI, the settings aren't recorded, and when I mount the SD card again, camera.txt files doesn't exist anymore. Client OS is macOS.

Any thoughts?

htruong, 2020-10-04:
Hey @Fernyman79 that doesn't sound right. Are you sure you're running https://github.com/showmewebcam/showmewebcam/releases/tag/V1.30-preview ? Can you show me what it says in:

```bash
cat /opt/uvc-webcam/start-webcam.sh
```

While you're logged to the ttyACM interface?

htruong, 2020-10-04:
Let's try the new preview version, it'll have more debugging info:

https://github.com/showmewebcam/showmewebcam/releases/tag/v1.30-preview2

If your settings are not applied correctly, try to issue this command on the ttyACM debug interface and paste here:

```bash
journalctl -t piwebcam
```

Fernyman79, 2020-10-04:
I got it to work. I was using the raspberry pi imager to write the SD card. Tried with etcher and it worked. @htruong your comment about my card not being written correctly tipped me. Thanks for the help!

Now that this is working, I have noticed that the camera framing is pretty close in/zoomed in. I'll open another issue, sine it's unrelated.

closed 2020-10-04 by htruong

htruong, 2020-10-04:
Thanks, everyone for testing and reporting bugs!

@PSF1 now we can handle the spaces in the settings file.

V1.30 is identical to V1.30-preview2.

commit 02a9f6a9 (2020-10-02): Piwebcam: Enable overriding of camera settings

Also, move the uvc-webcam script to buildroot dir so that we have better
separation between uvc-webcam and its startup logic.

How to override camera settings:

Connect to the ttyACM device and issue
/usr/bin/v4l2-ctl -L | less

Then, create a file called camera.txt in /boot and put all parameters
you want overridden, for example:

auto_exposure_bias=15

Resolves #2
commit 06269161 (2020-10-04): Piwebcam: Enable overriding of camera settings

Also, move the uvc-webcam script to buildroot dir so that we have better
separation between uvc-webcam and its startup logic.

How to override camera settings:

Connect to the ttyACM device and issue
/usr/bin/v4l2-ctl -L | less

Then, create a file called camera.txt in /boot and put all parameters
you want overridden, for example:

auto_exposure_bias=15

Resolves #2
commit d3d0cee9 (2020-10-04): Piwebcam: Enable overriding of camera settings

Also, move the uvc-webcam script to buildroot dir so that we have better
separation between uvc-webcam and its startup logic.

How to override camera settings:

Connect to the ttyACM device and issue
/usr/bin/v4l2-ctl -L | less

Note parameters that you want customized.

Then, mount the SD card on your computer, and create a file called
camera.txt in /boot and put all parameters you want overridden, e.g:

auto_exposure_bias=15

You can edit camera.txt on-target by remounting /boot read-write:

mount -o remount,rw /boot

Resolves #2