showmewebcam/showmewebcam #3: Poor performance on Cisco Webex (macOS Catalina)

Fernyman79, 2020-09-10:
Even thought the perforce of the camera is excellent in FaceTime or capturing videos from QuickTime, when I try to use it on Webex it falls drastically. Lots of dropped frames (only id say it only captures at 2-4 fps). Also, the quality of the video on the other end of the call is terrible.

I also noticed that Skype won't let me set this device as the default camera. When I select it from the menu the preferences window crashes.

Any idea how I can troubleshoot this?

htruong, 2020-09-11:
I think it's because the webcam only supports MJPEG for now. We need to get H264 support in for better performance. We need someone who can figure out how to make `uvc-camera` output H264 from the HQ camera.

```
$ ffmpeg -hide_banner -f v4l2 -list_formats all -i /dev/video0
[video4linux2,v4l2 @ 0x55b219f916c0] Compressed:       mjpeg :          Motion-JPEG : 1920x1080
```

kmalinich, 2020-09-12:
It may have something to do with whatever is telling OBS (here shown on macOS 10.15.6) that the only valid FPS selection is 2

[image]

htruong, 2020-09-13:
Thanks for the cursor. Let me see what can be done about this.

kmalinich, 2020-09-14:
Found it

https://github.com/showmewebcam/uvc-gadget/blob/4de8718f04d3fb14cee7fd1e3ceca0c93dc580d6/multi-gadget.sh#L29

Frame interval in was set to 5000000 microseconds, which is 2 frames per second.
Setting it instead to 333333 microseconds `[ (1/30) x 10^7) ]` makes it 30 frames per second.
Previous to this change, it did definitely appear to be running faster than 2 frames per second in Quicktime Player movie recording and in OBS preview, but now it is definitely much smoother. It may or may not fix @Fernyman79 's issue.

With this change the FPS value appears properly in OBS:
[image]

htruong, 2020-09-14:
@kmalinich Thanks, that's fantastic, I will try that and release a fix. I think the unit for `dwFrameInterval` isn't microseconds though,  because 333,333 microsecs = 0.33 secs = 3fps.

htruong, 2020-09-14:
I have checked, it looks like at least VLC in Linux sees that the current fps is 0.2.

That makes `dwFrameInterval` should be in microseconds, and thus be set to `33333` not `333333`.

Fernyman79, 2020-09-14:
Awesome! I'm definitely getting around 30 fps on Webex now. I can't test the image quality on the other end yet, but I'll report as soon as I have a chance to test.

htruong, 2020-09-14:
@Fernyman79 did you set the parameter `dwFrameInterval` to `33,333` or `333,333`?

nathcoad, 2020-09-14:
How can I modify the multi-gadget.sh script? The squashfs filesystem is mounted readonly. Thanks!

Fernyman79, 2020-09-14:
@htruong just checked. 333,333.

Fernyman79, 2020-09-14:
@nathcoad after editing the script, you have to rebuild:

`sudo make clean`
`sudo make`

I'm not sure if there's a proper way, but this worked for me.

kmalinich, 2020-09-14:
I've seen reference to the value being in nanoseconds in the UVC driver documentation, but still the value that makes 30fps happen is indeed 333333.

kmalinich, 2020-09-15:
Worked 100% for all (10+) the Teams calls I've used it with.

htruong, 2020-09-16:
Can someone help me test v1.22-preview that has the parameter set to 333,333?

https://github.com/showmewebcam/showmewebcam/releases/tag/v1.22-preview

nathcoad, 2020-09-16:
I don't have webex, but I do have OBS installed and I can confirm that the framerate has gone from 2 to 30 FPS, with a framerate interval of 33.3333 ms.
Thanks for fixing this so quickly!

PR #4 (2020-09-16): User/htruong/uprev piwebcam fps
Uprev piwebcam so we can get 30fps on cisco webex. Fixes #3.
closed 2020-09-16 by htruong