showmewebcam/showmewebcam #7: [Feature] IP camera

tamis-laan, 2020-09-25:
It would be awesome if the device could also function as an IP camera as wel as a webcam. 

We could expose a settings file through the usb connection, in which we could set the IP camera settings as wel as the ability to configure wifi settings.

htruong, 2020-10-02:
I will think about it. I will perhaps create another firmware to allow the Pi 0 to be used as an IP cam. I don't think it's favorable to do both.

tamis-laan, 2020-10-02:
I do see a use case for having both. The device can be used for video calling/work and at the same time function as a security device when the user is not home or on vacation. Of course the user should be able to turn both the webcam and IP camera functionality on or off to reduce overhead.

climberhunt, 2020-11-24:
I have an article on setting up as an IP camera with low latency here: http://www.davidhunt.ie/raspberry-pi-high-quality-camera-setup-for-low-latency-video-conferencing/ This was just before I did my article on the uvc-gadget solution (http://www.davidhunt.ie/raspberry-pi-zero-with-pi-camera-as-usb-webcam/). Both have their place, I needed a non-network solution, as my work VPN would not allow me to use an IP camera as a webcam for security reasons (network bridging).