showmewebcam/showmewebcam #9: Field of views problems

Fernyman79, 2020-10-04:
I'm using the raspberry pi v2 camera module. When testing the camera with raspistill, I get to see the full field of view. However, when I load the showmewebcam image on the Pi, the video is cropped. Some sample images:

> raspistill
[image]

> showmewebcam
[image]

I forgot to mention on my other thread: Thanks for making this! Trully awesome!

towolf, 2020-10-06:
Same here, it's really strongly cropped.

PSF1, 2020-10-07:
Same here. Pi 0w + 1.3 cam

towolf, 2020-10-07:
I think it's because of this table:

https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes

I would like 640x480, with 4x4 binning. Good quality and full view.

volfyd, 2020-10-07:
FYI, the modes for the HQ camera (and also V1 and V2) are here: https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

I wonder if you could get the mode you wanted by changing the values in https://github.com/showmewebcam/showmewebcam/blob/master/package/piwebcam/multi-gadget.sh

For example, s/1080/480/ and s/1920/640/

If that works, maybe the script could look for a resolution override file in /boot

htruong, 2020-10-07:
@towolf THanks for the excellent resource. I will take a look at it and see if changing the resolution solves the issue.

volfyd, 2020-10-07:
> FYI, the modes for the HQ camera (and also V1 and V2) are here: https://www.raspberrypi.org/documentation/raspbian/applications/camera.md
>
> I wonder if you could get the mode you wanted by changing [...]

For anyone trying this themselves, it seems that the resolution is also hard coded in uvc-gadget.c which is downloaded by the buildroot process somehow.

htruong, 2020-10-08:
UVC-gadget is pretty easy to deal with. The hard part is to (automatically) detect what kind of camera is connected to the raspberry pi so we can select the resolution accordingly. I am currently thinking we can use something in RPI_USERLAND package -- `raspistill -v` will print out what kind of camera we have, but it will increase boot time. 

I guess for the meantime, we can just specify it in `camera.txt`. Maybe something like this:

```
#camera_hardware=v1|v2|hq
```

htruong, 2020-10-08:
> I think it's because of this table:
> 
> https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes
> 
> I would like 640x480, with 4x4 binning. Good quality and full view.

Any reason you wanted 640x480 instead of 1296x972 given you have the V1 camera? 480p sounds very low res to me.

PSF1, 2020-10-08:
> ```
> #camera_hardware=v1|v2|hq
> ```

It maybe interesting to have an `auto` option too.
```
#camera_hardware=v1|v2|hq|auto
```

towolf, 2020-10-08:
> Any reason you wanted 640x480 instead of 1296x972 given you have the V1 camera? 640x480 sounds very low res to me.

No, I just picked that from the table, and it sounded like a reasonable standard resolution.

volfyd, 2020-10-08:
> UVC-gadget is pretty easy to deal with. The hard part is to (automatically) detect what kind of camera is connected to the raspberry pi so we can select the resolution accordingly. I am currently thinking we can use something in RPI_USERLAND package. `raspistill -v` will print out what kind of camera we have, but it will increase boot time. That is not very nice.
> 
> I guess for the meantime (while I'm try to figure out how to hack raspistill to just print out the camera name and exits right away), we can just specify it in `camera.txt`. Maybe something like this:
> 
> ```
> #camera_hardware=v1|v2|hq
> ```

It would be nice if you could also choose between 16:9 and 4:3. I prefer 16:9 but others might want 4:3 to get the full sensor.

towolf, 2020-10-08:
Yeah, I want the full view. The application can deal with the aspect ratio and the cropping.

htruong, 2020-10-08:
I tried to work on this yesterday and it turns out to be more problematic than I thought. I will put up a Work-in-progress PR here so someone who is smarter than me can chime in and see what's wrong with my code tonight...

volfyd, 2020-10-08:
> I tried to work on this yesterday and it turns out to be more problematic than I thought (got uvc-webcam crash for reasons I don't understand yet). I will put up a Work-in-progress PR here so someone who is smarter than me can chime in and see what's wrong with my code tonight...

I tried getting 2028x1520 on the HQ camera, and when I tried viewing the webcam stream the raspberry pi zero connection crashed including the serial connection. I tried increasing the value for /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1520p/dwMaxVideoFrameBufferSize (which oddly was set to 7372800 or 1920*1920*2), but increasing the value modestly didn't help.

htruong, 2020-10-08:
Yeah, that's what I experienced as well. What you can possibly do @volfyd if that helps you is to [enable wifi + SSH](https://github.com/showmewebcam/showmewebcam/issues/11#issuecomment-705317857) to debug further. I have been having not much luck yesterday even with Wifi (sometimes it connected, sometimes it didn't) maybe because my router is too far from the Pi, but you could try your luck. 

[Enabling ttyS1](https://github.com/showmewebcam/showmewebcam/blob/master/board/raspberrypi0cam/post-build.sh#L8) and use a USB-OTG cable and use a keyboard to login to the Pi directly can be another venue if you can't get wifi working.

I still haven't used the 40-pin debug header yet, it is supposed to work too.

weslebsack, 2020-10-19:
It would be awesome if this could get fixed. I installed my Pi0 inside of an Apple iSight enclosure, but I'm hoping there's an easy way to apply the update, since I'd have to disassemble the whole thing to get to the SD card. This is the only thing stopping me from having the ultimate webcam.

iliesaya, 2020-10-19:
can we change resolution to 1640x1232 ? would be nice to just have a line to change in boot/camera.txt

weslebsack, 2020-10-22:
Did this issue start on 1.3.0? Or is it on all previous versions?

PSF1, 2020-10-22:
1.21 is the first version that I install, and it have the problem.

iliesaya, 2020-10-22:
It is not an "issue" in a sense where this is how raspberry pi camera works : if you are not in 4k max resolution or 1640x1232 , then it is expected to only have a portion of the picture , which is a shame.

weslebsack, 2020-10-23:
> It is not an "issue" in a sense where this is how raspberry pi camera works : if you are not in 4k max resolution or 1640x1232 , then it is expected to only have a portion of the picture , which is a shame.

But it is an issue. It’s not just cropping the image—it’s digitally zooming **and** cropping. Take a look at the photos attached to the original post. The dog in the raw camera output appears further away than the one from showmewebcam. 

I’ve also used PiSight, which didn’t have this zooming/cropping problem. The reason I gave up on using PiSight was because it can’t be used when the MacBook Pro lid is closed, as it identifies as an Apple webcam.

tuyenld, 2020-11-01:
> But it is an issue. It’s not just cropping the image—it’s digitally zooming **and** cropping. Take a look at the photos attached to the original post. The dog in the raw camera output appears further away than the one from showmewebcam.

It is very strange. Today I tested with Pi Zero v1.3 + Camera v1, but it wasn't cropped and zoomed as the original post.
Here is the procedure:
- Write `sdcard-raspberry0-v1.40.img.gz` to SD card
- Create `/boot/camera.txt`
    ```bash
    auto_exposure_bias=15
    brightness=74
    ```
- Goto [https://webcamera.io/](https://webcamera.io/) to test image.
Here is the result:

Image with showmewebcam

[image]

Image with raspstill

[image]

I wonder if he views the image with the appropriate zoom ratio or not. 
As you can see, if two images have the same zoom ratio, there is not much difference.

[image]

[image]

Sorry for the poor image, I don't know how to adjust the color yet :((

towolf, 2020-11-01:
@tuyenld you do realize that your showmewebcam image has a much narrower field of view compared to your raspistill image?

Zoom is the wrong term for this,

weslebsack, 2020-11-01:
> @tuyenld you do realize that your showmewebcam image has a much narrower field of view compared to your raspistill image?
> 
> Zoom is the wrong term for this,

It's basically digital zoom. It's cropping a portion of the webcam image and enlarging it to the full resolution.

Here's the area that @tuyenld's camera is cropping:

[image]

And here's how it should be displaying:
[image]

[PiSight](https://github.com/maxbbraun/pisight) doesn't have this issue, so it's not a limitation with the camera. The problem with PiSight is it reports itself as an Apple FaceTime camera, which turns off when the MacBook Pro's lid is closed.

tuyenld, 2020-11-02:
@weslebsack thank you.
I got it. Because showmewebcam is running at 1920x1080 (16:9) so it is a partial FoV.
I am finding a way to config uvc-gadget.

tuyenld, 2020-11-04:
I found out that V4L2 hangs out at 2592x1944.
I test with `v4l2-ctl` on Official Raspberry Pi OS and it hangs out.
```bash
pi@raspberrypi:~$ v4l2-ctl --set-fmt-video=width=2592,height=1944,pixelformat=MJPG
pi@raspberrypi:~$ v4l2-ctl -p 30 #Initial frame rate, it may be changed
Frame rate set to 30.000 fps
pi@raspberrypi:~$ v4l2-ctl --stream-user=4 --stream-count=10 --stream-to=/dev/null  # Try with user pointer

^C
pi@raspberrypi:~$ v4l2-ctl --stream-mmap=4 --stream-count=10 --stream-to=/dev/null  # Try with mmap

^C
pi@raspberrypi:~$ 
pi@raspberrypi:~$ v4l2-ctl --set-fmt-video=width=2592,height=1944,pixelformat=l420
pi@raspberrypi:~$ v4l2-ctl -p 30
Frame rate set to 30.000 fps
pi@raspberrypi:~$ v4l2-ctl --stream-mmap=3 --stream-count=20 --stream-to=/dev/null
<<<<<< 4.15 fps
<<<< 4.04 fps
<<< 4.00 fps
<<<< 3.98 fps
<<<
pi@raspberrypi:~$ 
pi@raspberrypi:~$ v4l2-ctl --set-fmt-video=width=1440,height=1080,pixelformat=MJPG
pi@raspberrypi:~$ v4l2-ctl -p 30 #Initial frame rate, it may be changed
Frame rate set to 30.000 fps
pi@raspberrypi:~$ v4l2-ctl --stream-mmap=3 --stream-count=100 --stream-to=/dev/null
KKKKKKKKKKKKKKKKKKKKKKKKKKKKKKKK 30.02 fps
KKKKKKKKKKKKKKKKKKKKKKKKKKKKKK 30.02 fps
KKKKKKKKKKKKKKKKKKKKKKKKKKKKKK 30.02 fps
KKKKKKKK
pi@raspberrypi:~$ 

```
Thought `2592x1944` works with l420 at 4 fps (not works with JPEG), but I don't think it is suitable for transfer via USB.
`1440x1080 at 30fps` is optimal resolution for all camera version.

weslebsack, 2020-11-05:
> Though `2592x1944` works with l420 at 4 fps (not works with MJPG), but I don't think it is suitable for transfer via USB.
> `1440x1080 at 30fps` is optimal resolution for all camera version with **full FoV**.

@tuyenld does that mean you got it working at that resolution with full FoV?

tuyenld, 2020-11-06:
Yes, it is full FoV.
However, I just noticed today that the picture has a vertical blank pixel on the right.

[image]

I think 1440x1080 is full FoV because `2592/1440 = 1944/1080 = 1.8`
I may be miscalculating somewhere.

Jackbennett, 2020-11-07:
Is this configurable at boot somehow? I can't remount /opt to edit webcam.sh to even test. The device is busy to run set-fmt-video and killing ucv drops the serial connection.

`--set-fmt-video` isn't a -c option so can't be done with camera.txt

tuyenld, 2020-11-08:
@Jackbennett,
I tested through GPIO serial connection, so I didn't lose connection when uvc app was killed.
`roofs` partition is squashfs, therefore you can't edit `/opt`. I have to build a new image for testing.

htruong is planning to move to a new uvc-gadget repo which is able to reconfigure resolution without edit `camera.txt`.
Here is PR #28

weslebsack, 2020-11-13:
I'm assuming this issue would be closed if it was fixed, but is a fix for this included in 1.5.0?

tuyenld, 2020-11-14:
@weslebsack,
There are still some problems:
1. Several resolutions has a wide stripe line (black or white) on the right of the image (see my above image)
2. Some resolution need to be added to support full FoV such as 1640x922 for Cam v2, 1012x760 for HQ (currently, I believe only Camera v1 has full FoV at 640x480)

towolf, 2020-11-18:
With the current version 1.5.0 I get the full field of view.

And now I'm not sure If I like this better or not ... :feelshidingman:

weslebsack, 2020-11-18:
> With the current version 1.5.0 I get the full field of view.
> 
> And now I'm not sure If I like this better or not ... :feelshidingman:

Oh wow, if it's fixed in 1.5.0, that's awesome. Before I disassemble my camera to update the firmware, did you have to apply any settings in camera.txt? Or did it just automatically have the full FOV? Was it cropped in a 16:9 aspect ratio?

Jackbennett, 2020-11-19:
In my haste to give it a go I forgot to grab a before/after but that seems improved. I have had to add `auto_exposure_bias=12` to camera.txt to make it not so dark on 1.4 and 1.5. Which strikes me as odd since 12 is allegedly the default anyway.  Thanks to everyone involved this works really slick.

htruong, 2020-11-22:
Somehow at some point in the past I thought the lower auto-exposure setting is better. We'll fix it in https://github.com/showmewebcam/showmewebcam/pull/41 and hopefully will cut a release soon!

The field of view changes when you use OBS and change the camera resolution as expected. I think we can close this issue then?

weslebsack, 2020-11-25:
> The field of view changes when you use OBS and change the camera resolution as expected. I think we can close this issue then?

What's weird is, the field of view problem is fixed when I use the camera on macOS Big Sur, but it isn't fixed on macOS Catalina. Why would the OS make a difference in the field of view?

towolf, 2020-11-25:
They are requesting different image resolutions?

tuyenld, 2020-11-26:
> They are requesting different image resolutions?

I think so. V1.5 Showmewebcam image resolution completely depends on the application request. There is no "default resolution." Different applications request different resolutions, so they have different FoV.

As I remembered: 
- Zoom+Showmewebcam operate at 1280x720
- Skype+Showmewebcam operate at 640x480

JamesSwift, 2021-01-30:
I'm on v1.70 with the v2 camera module. The modes listed by the pi on the host computer don't seem to match the official list from `https://www.raspberrypi.org/documentation/hardware/camera/`, and the cropping/zooming of the sensor between different modes varies. For example, 1280x720 provides a full field of view, but 1920x1080 is "zoomed" significantly. At first I thought it was a 16:9 sensor being cropped to 4:3 then back to 16:9 (which would produce that zoomed effect), but the sensor is native 4:3, so I'm at a loss as to the process which produces these "zoomed images". See my attached examples.

Is there any way to always use the full resolution of the sensor for the requested framerate then get the pi's GPU to crop and down-sample to the requested size before sending to the host? This might also help with the softening of the image at lower requested resolutions that I have observed.

1280x720:
[image]

1280x960:
[image]

1920x1080:
[image]

smartroad, 2021-02-21:
@JamesSwift I have just started to use this and noticed the exact same thing. Also the 1080p image on both my cameras (ZeroCam and ZeroCam Fisheye) the video is much noisier than at 720p.

mattkerrison, 2021-07-01:
Can we get support for 1640x1232 resolution? on the V2 camera it's the best option for full fov:

[image]

leoshmu, 2021-07-12:
I'm getting same behavior as @JamesSwift - 1280 gives me the correct field of view, but 1920 gives me smaller field of view. I have to scale the 1920 image obtained by showmewebcam by about 1.75 or 1.8 to have the image align with the same scene captured in mode 1 (1920x1080 native) on a v2 camera with a csi interface . @tuyenld have you continued to work on this issue, curious if you have any further insights!

leoshmu, 2021-07-13:
> I'm on v1.70 with the v2 camera module. The modes listed by the pi on the host computer don't seem to match the official list from `https://www.raspberrypi.org/documentation/hardware/camera/`, and [...]

Interestingly, running image coregistration between the cropped image and the full FOV image, there is a 2.56 factor scaling and then the images are perfectly aligned. Can't figure out where that 2.56 come from. 
[image]

leoshmu, 2021-07-13:
> I'm getting same behavior as @JamesSwift - 1280 gives me the correct field of view, but 1920 gives me smaller field of view. I have to scale the 1920 image obtained by showmewebcam by about 1.75 or [...]

Update - I found that I was making a mistake. I was using the PiCamera class to capture an image with 1920x1080 resolution, but I was not setting the sensor mode explicitly. As a result, the camera was giving me an image captured in sensor mode 0 (3280x2464) and then resizing it to 1920x1080. This made it appear to me that the showmewebcam was capturing a different field of view than the csi connected pi camera. However, once I explicitly set the sensor mode to sensor_mode=1, then the PiCamera class obtained the 1920x1080 native resolution partial field of view, corresponding exactly to the field of view captured by the showmewebcam with 1920x1080 resolution. Thanks to the creators of this wonderful project and apologies for not realizing the error was on my end!

closed 2021-08-22 by dgsiegel