showmewebcam/showmewebcam #11: Won’t boot on pi zero w(h)

puccaso, 2020-10-07:
Hello, 
I really enjoyed your article however I am unable to get the image to boot. 

Any ideas?

[image]

Many thanks,

puc

puccaso, 2020-10-07:
Also, doing some further troubleshooting, 
I had initially used your latest image, so this time i went with the latest stable of buildroot and a git clone of your repo.. 

i currently get this error when trying to build both the latest stable and the LTS release

`tmp-com.s: Assembler messages:
tmp-com.s:65: Error: selected processor does not support `vld1.32 {d0[0]},[r1]!' in ARM mode
tmp-com.s:67: Error: selected processor does not support `vmvn d0,d0' in ARM mode
tmp-com.s:68: Error: selected processor does not support `vst1.32 {d0[0]},[r0]!' in ARM mode
tmp-com.s:71: Error: selected processor does not support `vld1.32 {d0},[r1]!' in ARM mode
tmp-com.s:73: Error: selected processor does not support `vmvn d0,d0' in ARM mode
tmp-com.s:74: Error: selected processor does not support `vst1.32 {d0},[r0:64]!' in ARM mode
tmp-com.s:75: Error: selected processor does not support `vld1.32 {q2},[r1]!' in ARM mode
tmp-com.s:80: Error: selected processor does not support `vld1.32 {q0},[r1]!' in ARM mode
tmp-com.s:81: Error: selected processor does not support `vmvn q2,q2' in ARM mode
tmp-com.s:83: Error: selected processor does not support `vst1.32 {q2},[r0:128]!' in ARM mode
tmp-com.s:84: Error: selected processor does not support `vld1.32 {q2},[r1]!' in ARM mode
tmp-com.s:85: Error: selected processor does not support `vmvn q0,q0' in ARM mode
/bin/bash ../libtool  --tag=CC   --mode=compile /usr/lib/distcc/gcc -DHAVE_CONFIG_H -I. -I..  -D__GMP_WITHIN_GMP -I.. -DOPERATION_`echo sub_err2_n | sed 's/_$//'` -I/home/pi/Desktop/buildroot-2020.08/output/host/include  -O2 -I/home/pi/Desktop/buildroot-2020.08/output/host/include -c -o sub_err2_n.lo sub_err2_n.c
tmp-com.s:86: Error: selected processor does not support `vst1.32 {q0},[r0:128]!' in ARM mode
tmp-com.s:89: Error: selected processor does not support `vmvn q2,q2' in ARM mode
tmp-com.s:90: Error: selected processor does not support `vst1.32 {q2},[r0:128]!' in ARM mode
tmp-com.s:96: Error: selected processor does not support `vld1.32 {q0},[r1]!' in ARM mode
tmp-com.s:97: Error: selected processor does not support `vmvn q0,q0' in ARM mode
tmp-com.s:98: Error: selected processor does not support `vst1.32 {q0},[r0]!' in ARM mode
tmp-com.s:101: Error: selected processor does not support `vld1.32 {d0},[r1]!' in ARM mode
tmp-com.s:102: Error: selected processor does not support `vmvn d0,d0' in ARM mode
tmp-com.s:103: Error: selected processor does not support `vst1.32 {d0},[r0]!' in ARM mode
tmp-com.s:106: Error: selected processor does not support `vld1.32 {d0[0]},[r1]' in ARM mode
tmp-com.s:107: Error: selected processor does not support `vmvn d0,d0' in ARM mode
tmp-com.s:108: Error: selected processor does not support `vst1.32 {d0[0]},[r0]' in ARM mode
distcc[8209] ERROR: compile (null) on localhost failed
/bin/bash ../libtool  --tag=CC   --mode=compile /usr/lib/distcc/gcc -DHAVE_CONFIG_H -I. -I..  -D__GMP_WITHIN_GMP -I.. -DOPERATION_`echo sub_err3_n | sed 's/_$//'` -I/home/pi/Desktop/buildroot-2020.08/output/host/include  -O2 -I/home/pi/Desktop/buildroot-2020.08/output/host/include -c -o sub_err3_n.lo sub_err3_n.c
make[3]: *** [Makefile:768: com.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
libtool: compile:  /usr/lib/distcc/gcc -DHAVE_CONFIG_H -I. -I.. -D__GMP_WITHIN_GMP -I.. -DOPERATION_add_err1_n -I/home/pi/Desktop/buildroot-2020.08/output/host/include -O2 -I/home/pi/Desktop/buildroot-2020.08/output/host/include -c add_err1_n.c  -fPIC -DPIC -o .libs/add_err1_n.o
libtool: compile:  /usr/lib/distcc/gcc -DHAVE_CONFIG_H -I. -I.. -D__GMP_WITHIN_GMP -I.. -DOPERATION_add_err2_n -I/home/pi/Desktop/buildroot-2020.08/output/host/include -O2 -I/home/pi/Desktop/buildroot-2020.08/output/host/include -c add_err2_n.c  -fPIC -DPIC -o .libs/add_err2_n.o
libtool: compile:  /usr/lib/distcc/gcc -DHAVE_CONFIG_H -I. -I.. -D__GMP_WITHIN_GMP -I.. -DOPERATION_add_err3_n -I/home/pi/Desktop/buildroot-2020.08/output/host/include -O2 -I/home/pi/Desktop/buildroot-2020.08/output/host/include -c add_err3_n.c  -fPIC -DPIC -o .libs/add_err3_n.o
libtool: compile:  /usr/lib/distcc/gcc -DHAVE_CONFIG_H -I. -I.. -D__GMP_WITHIN_GMP -I.. -DOPERATION_sub_err1_n -I/home/pi/Desktop/buildroot-2020.08/output/host/include -O2 -I/home/pi/Desktop/buildroot-2020.08/output/host/include -c sub_err1_n.c  -fPIC -DPIC -o .libs/sub_err1_n.o
libtool: compile:  /usr/lib/distcc/gcc -DHAVE_CONFIG_H -I. -I.. -D__GMP_WITHIN_GMP -I.. -DOPERATION_sub_err3_n -I/home/pi/Desktop/buildroot-2020.08/output/host/include -O2 -I/home/pi/Desktop/buildroot-2020.08/output/host/include -c sub_err3_n.c  -fPIC -DPIC -o .libs/sub_err3_n.o
libtool: compile:  /usr/lib/distcc/gcc -DHAVE_CONFIG_H -I. -I.. -D__GMP_WITHIN_GMP -I.. -DOPERATION_sub_err2_n -I/home/pi/Desktop/buildroot-2020.08/output/host/include -O2 -I/home/pi/Desktop/buildroot-2020.08/output/host/include -c sub_err2_n.c  -fPIC -DPIC -o .libs/sub_err2_n.o
make[3]: Leaving directory '/home/pi/Desktop/buildroot-2020.08/output/build/host-gmp-6.1.2/mpn'
make[2]: *** [Makefile:956: all-recursive] Error 1
make[2]: Leaving directory '/home/pi/Desktop/buildroot-2020.08/output/build/host-gmp-6.1.2'
make[1]: *** [Makefile:775: all] Error 2
make[1]: Leaving directory '/home/pi/Desktop/buildroot-2020.08/output/build/host-gmp-6.1.2'
make: *** [package/pkg-generic.mk:250: /home/pi/Desktop/buildroot-2020.08/output/build/host-gmp-6.1.2/.stamp_built] Error 2
pi@flux:~/Desktop/showmewebcam$ 
`

i've tried building on a raspberry pi 4 8gb, and 4gb, running latest stable of raspberry pi os with buster. 
i've tried with and without distcc and it exits the process at the same location.

htruong, 2020-10-08:
@puccaso I don't know about your second issue. I have been always building on my Ubuntu 20.04 laptop and it worked fine, perhaps buildroot doesn't support building on ARM very well.

For the first issue, you can try to do this:

- Go to `showmewebcam` folder.
- Revert commit `5de44f96bf904269c2548efb3407436e8ad4fac5`, you'll have dropbear and wifi on your pi, remember to edit `board/raspberrypi0cam/rootfs/etc/wpa_supplicant/wpa_supplicant-wlan0.conf` so it matches your config and don't steal my top secret wifi.

Then,
- Go to buildroot folder.
- `BR2_EXTERNAL=../showmewebcam/ make raspberry0cam_defconfig`
- `rm -rf output`
- `make all`

Hopefully you'll see the pi on your wireless network with a dropbear interface.

htruong, 2020-10-08:
Oh before you try that, for the `ttyACM0` issue, please first do `sudo dmesg -w` and plug your pi to the computer, then watch `dmesg` output. Perhaps your Pi doesn't appear as `ttyACM0`, maybe it's `ttyACM1` or something weird.

htruong, 2020-10-08:
If you see that ttyACM0/1 shows up, but you see it disconnected right afterwards in dmesg, then it's very likely that uvc-gadget failed to start. You should check the camera or the cable to the camera, likely it's bad.

puccaso, 2020-10-09:
Firstly, thank you so much for your time and attention.

Seems like I have a getty issue, 

`Oct  9 20:58:31 flux agetty[1196]: /dev/ttyGS0: cannot open as standard input: No such file or directory`

I will resolve and post back.

puccaso, 2020-10-09:
Ok, some progress. 
I can get both macos and my other rpi to see the device now in dmesg..

`[  791.059337] usb 1-1.3: new full-speed USB device number 9 using xhci_hcd
[  791.165279] usb 1-1.3: New USB device found, idVendor=0a5c, idProduct=2763, bcdDevice= 0.00
[  791.165296] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  791.165309] usb 1-1.3: Product: BCM2708 Boot
[  791.165321] usb 1-1.3: Manufacturer: Broadcom
`

is this normal? What is Product: BCM2708 Boot? is that the /boot partition?

puccaso, 2020-10-11:
And we're good!

Thank you so much for your help. 
finally working on BigSur =]

`
Show_me_webcam Pi Webcam:

  Model ID:	UVC Camera VendorID_7531 ProductID_260
  Unique ID:	0x14100000********

`

You Rock!

closed 2020-10-11 by puccaso