showmewebcam/showmewebcam #14: Serial interface on 40pin header requires enable_uart=1 in config.txt

ranma, 2020-10-15:
> Also, there is a untested serial interface on the serial 40-pin header: https://pinout.xyz/pinout/uart

It's not working with the default image.
To enable the serial port "enable_uart=1" needs to be added to /boot/config.txt.

After that both boot logs and logging in are working:
```
[    0.000000] Booting Linux on physical CPU 0x0                             
[    0.000000] Linux version 4.19.113 (wasabi@anna) (gcc version 8.4.0 (Buildroot 2020.02.3)) #1 Sun Jul 26 10:01:21 PDT 2020
[...]
Welcome to Buildroot 2020.02.3!

[    2.586816] systemd[1]: Set hostname to <showmepicam>.
[...]
```

htruong, 2020-10-19:
Thanks, I will fix it in the next release!

htruong, 2020-10-24:
The change is merged, I will put up a new build shortly with the change.

closed 2020-10-24 by htruong