showmewebcam/showmewebcam #15: [Feature] Buttons on GPIO header

ranma, 2020-10-16:
I would be nice to have a a default setup for some simple buttons on the GPIO header to control some of the camera functions.
e.g. a button to disable/enable auto-exposure or disable/enable the video feed.

It could also be interesting to use usb_f_hid and expose buttons to the host via HID (e.g. a hangup button).

rcarmo, 2020-10-25:
I am looking at using one of the Adafruit mini-LED displays (with buttons) to build a simple way to do that. Waiting for the hardware to arrive.

gee33, 2021-01-10:
👍🏼

neherh, 2021-02-25:
@rcarmo any more news on this front?

rcarmo, 2021-02-25:
I got the display, but didn't yet find the time to design and 3D print a suitable enclosure or code some UI.