showmewebcam/showmewebcam #19: Error building for raspberry Pi Zero (without wifi)

joanArchguy, 2020-10-24:
I followed the steps in the readme for building an image for a raspberry pi zero (without wifi) and at some point a got this error:

`/usr/bin/ld: scripts/dtc/dtc-parser.tab.o:(.bss+0x10): multiple definition of 'yylloc'; scripts/dtc/dtc-lexer.lex.o:(.bss+0x0): first defined here`
`collect2: error: ld returned 1 exit status`
`make[3]: *** [scripts/Makefile.host:99: scripts/dtc/dtc] Error 1`
`make[2]: *** [scripts/Makefile.build:544: scripts/dtc] Error 2`
`make[2]: *** Waiting for unfinished jobs...`
`  HOSTLD  scripts/mod/modpost`
`make[1]: *** [Makefile:1068: scripts] Error 2`
`make[1]: *** Waiting for unfinished jobs....`
`make[1]: Leaving directory '/home/develop/buildroot/output/build/linux-custom'`
`make: *** [package/pkg-generic.mk:269: /home/develop/buildroot/output/build/linux-custom/.stamp_built] Error 2`

Does anyone have any clue on what could be causing it, or a working image for a raspberry pi zero (without wifi)?

htruong, 2020-10-25:
Please try my repo to see if ti works, I just built successfully with it from scratch.

https://github.com/htruong/showmewebcam/tree/user/htruong/fix_build_script

I will post a binary image shortly so you can try too.

htruong, 2020-10-25:
Can you help testing this image whenever you have a chance?

https://github.com/showmewebcam/showmewebcam/releases/tag/v1.40-preview

joanArchguy, 2020-10-25:
> ```shell
> BUILDROOT_DIR=../buildroot ./build-showmewebcam.sh raspberrypi0
> ```

joanArchguy, 2020-10-25:
> 
> 
> Please try my repo to see if ti works, I just built successfully with it from scratch.
> 
> ```shell
> BUILDROOT_DIR=../buildroot ./build-showmewebcam.sh raspberrypi0
> ```
> 
> https://github.com/htruong/showmewebcam/tree/user/htruong/fix_build_script
> 
> I will post a binary image shortly so you can try too.
> 
> NOTE: The way you invoke the script has changed, you have to tell the script where the buildroot dir is, and the output dir will be in the showmewebcam directory as well.

I have been able to build successfully with this branch, thank you.

joanArchguy, 2020-10-25:
> 
> 
> Can you help testing this image whenever you have a chance?
> 
> https://github.com/showmewebcam/showmewebcam/releases/tag/v1.40-preview

This is working perfectly as long as I tested, today I plan to make a long video call with it and also test its stability a little bit.
Is there any specific test you want me to try? I would love to be able to help to continue growing this amazing project.

htruong, 2020-10-25:
Yeah if it works and boots up, I think it should be fine. I actually don't have a Pi0 (non W) to test. :) Thanks and thanks for reporting that the patch works!

closed 2020-10-25 by htruong

joanArchguy, 2020-10-26:
> This is working perfectly as long as I tested, today I plan to make a long video call with it and also test its stability a little bit.
> Is there any specific test you want me to try? I would love to [...]

Hi! Yesterday I was using the Raspberry Pi Zero (without WiFi) with the HQ cam module and the v1.40-preview version burned in during a Skype video call. At the beginning it was working perfectly but slowly, and after 5 minutes the image started showing a kind of horizontal lines (as in the attached capture) and the lines kept being more abundant as time passes, until the system  froze and crashed, leaving my computer (Windows 10) uncapable of recognize the USB Cam.
[image]

tuyenld, 2020-10-26:
Hi @joanArchguy ,
I tried a quick test. This is my setting:
- SW: sdcard-raspberry0-v1.40.img.gz
- HW: Pi Zero v1.3 (no Wi-Fi) + camera v1 5MP
- Skype call from Window 7 to my iPhone for 30 minutes
I didn't encounter your mentioned issue. My camera worked fine. I don't have HQ camera to test right now.

joanArchguy, 2020-10-26:
Hi @tuyenld thanks for your quick test! So it could be my HQ cam module that is been damaged. Anyways is there any logs I can look into?

htruong, 2020-10-26:
@joanArchguy 

Three things that could be wrong:
- The Pi
- The cable
- The camera

If you have more than a Pi, then you can do some troubleshooting by swapping them to the working Pi, to see if the Cable+Camera is the problem. It may be that your cable is bad. Was it bent? I had the bad cable syndrome before and the syndrome was like yours. Not saying it's your case here but something to look into.

joanArchguy, 2020-10-26:
I checked the cable and it was appreciably bent, I flattened it as much as I can, but if it's the cable I assume the damage is irreversible, so I will borrow another Pi to confirm that the problem is cable + cam. Thank you for your advice I really appreciate it.