showmewebcam/showmewebcam #21: Design a simple laser cut sandwich piece for the HQ camera and Pi0

htruong, 2020-10-25:
Would any kind soul with some CAD skill design a acrylic piece to sandwich between the camera and the raspberry pi board so they don't move? I am so tired of having to hold this thing.

Requirements:
- Has 4 holes to hold the pi and 4 holes to hold the HQ camera. Camera is in the front of the Pi like picture attached. Works for the short ribbon cable.
- Allows the HQ camera to be mounted with a camera tripod piece, so the Pi cannot be too low because that will bend the flex cable.

[image]

(If you have one and can send my way, that'd be extra appreciated!)

miip, 2020-10-25:
I 3d printed this https://www.thingiverse.com/thing:4445767

Works great

dgsiegel, 2020-10-25:
Or buy this one: https://thepihut.com/products/mounting-plate-for-high-quality-camera

htruong, 2020-10-25:
Thanks @dgsiegel I didn't know that they offer this piece for sale, it's exactly what I wanted!

closed 2020-10-25 by htruong