showmewebcam/showmewebcam #23: Add camera.txt to /boot during build

dgsiegel, 2020-11-03:
It would be awesome if an existing camera.txt could be copied to /boot when building an image yourself. With that in place, we could also remove 

```
/usr/bin/v4l2-ctl -c auto_exposure_bias=3
/usr/bin/v4l2-ctl -c video_bitrate=25000000
```
from `package/piwebcam/start-webcam.sh`. The only solution I found would be to add the camera.txt file to `board/raspberrypi0cam/genimage-raspberrypi0.cfg` and add these lines to `board/raspberrypi0cam/post-image.sh`:

```
# Copy camera.txt if it exists in build dir
if [ -f "${BR2_EXTERNAL_PICAM_PATH}/camera.txt" ]; then
  cp "${BR2_EXTERNAL_PICAM_PATH}/camera.txt" "${BINARIES_DIR}/rpi-firmware/camera.txt"
fi
```

But that requires an existing camera.txt in the root folder or the build will fail.

htruong, 2020-11-04:
Not that I tested this idea, but we could put a dummy blank `camera.txt` file in the repo so that it doesn't upset the `cfg` file config, and then somehow dynamically swap it to the real file to it during build, maybe?

dgsiegel, 2020-11-04:
That would work if we add the above mentioned camera.txt file to the repo. Don't know if that is the best way though...

PSF1, 2020-11-06:
The repo has a camera.txt.example in it. That file have configuration details but it don´t change default values. it maybe copied how a empty camera.txt file while building.

dgsiegel, 2020-11-06:
Well, we already have these two lines in `package/piwebcam/start-webcam.sh`:
```

/usr/bin/v4l2-ctl -c auto_exposure_bias=3
/usr/bin/v4l2-ctl -c video_bitrate=25000000
```

They could easily go into the default camera.txt, no?

dgsiegel, 2020-11-10:
Here's a pull request for this: https://github.com/showmewebcam/showmewebcam/pull/31

rcarmo, 2020-11-10:
It would be nice to have a sample `camera.txt` by default with the most common settings there (including rotation, which is what I need to tweak on every new build...).

dgsiegel, 2020-11-10:
Yeah, we have some settings already (https://github.com/showmewebcam/showmewebcam/issues/23#issuecomment-723344994). What other settings would you propose?

Regarding rotation, that probably depends on how you use your mount?

rcarmo, 2020-11-10:
I don’t have a standard mount (I built my own compact mount, which is depicted in https://taoofmac.com/space/blog/2020/10/11/1800 <https://taoofmac.com/space/blog/2020/10/11/1800> and requires me to set the rotation to 270).

I would say that brightness, contrast and quality/bitrate would be among the top 5 things to have available by default...

dgsiegel, 2020-11-10:
Oh ok. I guess not that many people will have their camera mounted at 270 degrees :) Do you build the image yourself? In either case you can set defaults, which won't get overwritten.

Brightness and contrast are set to auto right now. The quality/bitrate to max as seen above.

rcarmo, 2020-11-10:
No, I just shove a camera.txt in after flashing the image. Would be nice to have one there, though. Quicker to edit :)

dgsiegel, 2020-11-10:
Well with my PR you can have your camera.txt in your showmewebcam directory and have it copied over automatically. I can't really think of a solution when flashing the image though...

rcarmo, 2020-11-10:
Nah, I just want a set of sensible defaults inside the image. I just flashed the latest, and even the .sample file would have come in handy in there.

dgsiegel, 2020-11-11:
Alright. Could you try the following settings? I've tried them locally and they seem to provide a good, average result:

```
auto_exposure=0
auto_exposure_bias=12
contrast=-2
saturation=5
video_bitrate=25000000
sharpness=20
```

dgsiegel, 2020-11-19:
@htruong would you mind trying these defaults on your 6mm lens?

htruong, 2020-11-22:
My picture with the default settings on a bright day:

[image]

I actually have my camera.txt to be so:

```
auto_exposure_bias=15
contrast=0
brightness=55
```

Then my picture is way better:

[image]

What do you think about this setting?

htruong, 2020-11-22:
Your settings:

[image]

htruong, 2020-11-22:
Let's use yours as default.

closed 2020-11-22 by dgsiegel