showmewebcam/showmewebcam #30: [Feature] Enable GPIO pin on webcam stream

Leonti, 2020-11-08:
One feature that was missing for me was the lack of LED to indicate that the camera is streaming.
I've done it locally and it works well, here is the picture of the camera when streaming:
[image]

I've modified my local copy of https://github.com/showmewebcam/uvc-gadget to do that (enable pin 21 when streaming starts and disable when streaming ends) and was about to make a PR, but it seems like the project will be moving to https://github.com/peterbay/uvc-gadget soon.  

Which repo would be best to submit a PR?

peterbay, 2020-11-08:
Put your local copy to github and i will implement it to my version of uvc-gadget

peterbay, 2020-11-08:
I've created room at discord for discussion about uvc-gadget - https://discord.gg/EN7AJNw5

Leonti, 2020-11-08:
Hi @peterbay!
That would be awesome!

Here is the code I'm running at the moment:
https://github.com/showmewebcam/uvc-gadget/compare/master...Leonti:gpio-21?expand=1

dgsiegel, 2020-11-09:
Seems like this was done https://github.com/peterbay/uvc-gadget/commit/bf1b5fba283d1b08f8616a9d3c1567b4eeca9d20. It also works with the onboard led: https://github.com/peterbay/uvc-gadget/commit/ad13e70614433327be74b7f19fb7774f8cb9d08d

htruong, 2020-11-22:
I'm closing this issue due to its spirit being implemented. :)

closed 2020-11-22 by htruong