showmewebcam/showmewebcam #34: [HELP] Issues adding camera.txt to /boot.

Maevarienn, 2020-11-12:
First off, I apologize for what I believe to be a classic case of user error and feel free to ignore/close this whenever, but I seem to be unable to add a camera.txt or anything to the boot directory of the image file. I am not that well versed in Linux, but remounting the partition as read/write seems to fail saying that the directory is marked as read-only and cannot be changed. I am using Ubuntu 20.10 to attempt this and have tried many methods of writing to the directory, but I cannot seem to get it out of a read-only state. Again, sorry for this "issue", but I really have no clue what I'm doing wrong, maybe you fine folks here could provide some input? Thanks for the amazing project and any help provided in advance!

tuyenld, 2020-11-12:
Hi @Maevarienn,

I checked with the latest image v1.5.0 and it still works. 
That is what I did:
```bash
# ## Check the sd-card mounted or not
# mount | grep mmc
/dev/mmcblk0p2 on / type squashfs (ro,relatime)
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
# 
# mount -o remount,rw /boot
# 
# mount | grep mmc
/dev/mmcblk0p2 on / type squashfs (ro,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
# ## Now you can write camera.txt in to /boot directory
```
Another way to create the `camera.txt` is that you unplug SD-card from your Raspberry Pi and plug it into the PC.
You can create and edit it directly from your computer like this:

[image]

Maevarienn, 2020-11-13:
Thank you so much for the help, the attached photos are very clear in telling me that I was attempting to place the file in the wrong spot. Thanks again for that, I'm glad I got that fixed I was really confused on what I was doing incorrect. Bah, of course it was something so simple.

closed 2020-11-13 by Maevarienn