showmewebcam/showmewebcam #37: Serial console output is garbled after boot

dgsiegel, 2020-11-16:
When connecting to the pi after a reboot, the serial console output seems to be garbled. When connecting again, everything seems to be fine. You can press enter twice and then log in without problems.

```
# picocom -b 115200 /dev/ttyACM0
picocom v3.1

port is        : /dev/ttyACM0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
login: 

       piwebcam login: 

                       piwebcam login: 

                                       piwebcam login: 

                                                       piwebcam login: 

                                                                       piwebcam login: 

                                                                                       piwebcam login: Your webcam at
Password: 

```

htruong, 2020-11-24:
I have never encountered that. What did you do specifically?

dgsiegel, 2020-11-24:
I just connected with picocom. It always happens on a fresh boot. Once logged in, the issue disappears.

htruong, 2020-12-20:
I know what happened, the way we set up the usb gadget made it garbled. If you disable webcam function, it will no longer get garbled. It seems like for whatever reason, the device that initializes first in the `multi-gadget.sh` (soon to be `usb-gadget-config.sh`) will work fine, but the second and the third function will be garbled.

matiaspl, 2021-01-04:
I think PiKVM uses the composite mode (mass storage along with HID in particular). Maybe there's some worthy knowledge there? I even think there might be mutual profit if PiKVM gains webcam support (either virtual, say mjpeg stream or from the camera module).

akseidel, 2021-02-07:
I have seen the same sort of output in OS X in a Terminal screen session. I assume by the looks it is the same problem. Issuing a 'reset' command in Terminal prevents the problem.

akseidel, 2021-02-19:
I made a video placed privately on youtube that demonstrates this issue in a number of ways. The way the issue manifests varies from time to time. Some clues are subtle, like remaining parts of the greeting text lines arriving at different moments. I have become convinced it is the root cause for almost all the problems I have encountered automating the serial console task. The video might also be showing some things one might not notice without having observed hundreds of serial console attempts where one begins to notice slight operation variations.  

[https://www.youtube.com/watch?v=JsnuBY1NZbQ](url)

dgsiegel, 2021-02-24:
Can this be resolved with #109 ?

akseidel, 2021-03-01:
Running a build having the autolog changes in #109 considerably improves the situation when using the picamctl.sh script but it looks like remnants of what I guess is the underlying cause for the garble shows up when running picamctl.sh using the manual log argument. The resulting serial 'screen' connection typically shows 12 or so lines of # where the last line is # # space. The system is logged in as root but any thing typed in at the prompt is actually # space <what you type>. On a Pi4 there are multiple sequences of 4 lines of # where the every fourth line is # # space. I never saw any picamctl.sh misfires when it logs in. The 'root' 'root' it sends for logging in as root serves to clear the '# space' trash that would have interfered with the 'camera-ctl' send. In other words removing the 'root' 'root' send prior to the 'camera-ctl' send can result with a '# space camera-ctl' instead of a clean 'camera-ctl' entry.