showmewebcam/showmewebcam #43: Microphone support?

mattkerrison, 2020-11-23:
Any chance of supporting a Mic? there's a few cheap boards that would work with the GPIO ports on the Pi

htruong, 2020-11-24:
For a good setup, you'd want a dedicated microphone. A shabby microphone such as the one that is on a board on the Pi and far away from your face would require a lot of work to get rid of the noise while something [in this list](https://www.amazon.com/s?k=usb+microphone+clip+on) will serve you very well.

Not that I'm against the idea of supporting microphones eventually, but I think it will take some time to sort out while having a good webcam is always going to be the top priority. :)

StuartIanNaylor, 2020-12-08:
@htruong there is your i2s mic repo https://github.com/opencardev/snd-i2s_rpi

Maybe https://invensense.tdk.com/wp-content/uploads/2015/02/Microphone-Array-Beamforming.pdf might interest you.
You can create an endfire beamformer by just delaying by 2 samples and adding the inverse of the rear to the front mic.
The frequency dependent response can be fixed with alsaeq https://www.hifiberry.com/docs/software/guide-adding-equalization-using-alsaeq/ and you should have quite a decent quality directional mic sort of shotgun style?
https://www.hifiberry.com/docs/software/guide-adding-equalization-using-alsaeq/

@ 48khz the min mic spacing of 7mmm and overall spacing that creates the null notch needs some play but thinking it would be an excellent addition if it works?

ccarse, 2021-01-05:
+1 It seems like this would be a great addition to the project. This would allow us to sync the video track to the sound track more easily I'd think?

StuartIanNaylor, 2021-01-06:
Its sort of important to record sound as part of the process than as a separate file to be latency adjusted and lip synced after.

If the Pi4 ever comes to fruition then there are USB that can use just plain old USB soundcards.

The IQaudio Codec Zero hat is also another interesting product as it has 2x I2S mics and a mono electret input.
There are also extremely cheap  PCM1802 modules that as far as I know just need a slight modification to the https://github.com/opencardev/snd-i2s_rpi code.

As a product review I suggest giving the [boya-MM1](http://www.boya-mic.com/smartphonemicrophones/BY-MM1.html) a go as it can be found for £15 on ebay and from me and a rake of reviews on youtube its excellent for the $

htruong, 2021-01-06:
Something like this could work:

https://ceworkbench.wordpress.com/2019/07/21/setting-up-a-composite-ethernet-audio-usb-gadget/

I'm currently probably not going to have tons of time looking at it but this sounds plausible with some minimal effort. I do have a i2s mic and will try to work on it whenever it's convenient. Anyone else that knows alsa is also welcome to do so.

StuartIanNaylor, 2021-01-06:
Never thought of it as a composite driver as 1.1.x and 1.3.x provide support for MJPEG and H264 video formats as well so you gain compression as well.
I am a noob to the project and is raw YUY2 purely due to load on a Pi Zero?

jeztek, 2021-02-12:
Just created a [draft pull request](https://github.com/showmewebcam/showmewebcam/pull/102) that somewhat works.

I've tested it with an Infineon IM69D130 MEMS microphone + Analog Devices ADAU7002 PDM-to-i2s converter ([eval board](https://www.digikey.com/en/products/detail/infineon-technologies/S2GOMEMSMICIM69DTOBO1/9922560)) but any i2s microphone should suffice. 

It successfully produces audio from the microphone on the host device, however I'm seeing some odd behavior:
 * Using the camera + mic combo on Zoom works fine, but using any WebRTC service introduces artifacts in the video
 * On my Mac, I'm unable to simultaneously record audio and video using QuickTime Player for more than 1-3 seconds, however recording only audio or only video works fine
 * `alsaloop` CPU utilization is unusually high at ~30-40% when idle, but drops to 10-18% when the mic is in use
 * When I replace `alsaloop` with `speaker-test -D plughw:1,0 -r48000 -c2` I'm able to record audio and video in QuickTime with no problems
 
It seems like there's something going on between the uac2 Linux driver and alsaloop but I'm not sure how to proceed next. I tried setting `CONFIG_USB_DWC2_DEBUG=y` but it produced way too much output. Suggestions welcome!

taprobane99, 2021-02-15:
What about a dual mic Hat for the Pi Zero like this?

https://shop.pimoroni.com/products/respeaker-2-mics-phat

although might make the webcam a bit back-heavy with both a zero and this stacked on the back...

liquidbrains, 2021-06-19:
> What about a dual mic Hat for the Pi Zero like this?
> 
> https://shop.pimoroni.com/products/respeaker-2-mics-phat
> 
> although might make the webcam a bit back-heavy with both a zero and this stacked on the back...

I've got one on the back of my Pi Zero HQ cam combo and it is sitll front heavy (on the camera lens side).  I'm probably going to try and make a cover for the back so that it focuses the audio.  I might get some stand-off screws so that I can turn the boards around and have the mics more at the front and the cable out of the way.

[image]

StuartIanNaylor, 2021-06-19:
Its far from ideal as any extended ports are going to be essentially a bass tube.

What you can do is mount 2x I2s mics direct to gpio as adafruit https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-test

The aliexpress I2s mics are even cheaper https://www.aliexpress.com/item/32970296252.html

If you install pulseaudio and just the beamforming part of WebRTC AEC then at least you have a bit of beamforming and essentially the stereo pair becomes a single channel shotgun mic as also with the 2 mic hats having 2x onmi-directional  channels with a 58mm spacing is a whole load of pointless.