showmewebcam/showmewebcam #46: Build 1.5.0 questions

puccaso, 2020-11-26:
Hello, 

Great work!

3 questions:
- does the 1.5.0 build work on a pi4?
- how do we access the wifi functionality for the pi0w build? i'm on the device via serial but ifconfig wifi shows no such device..
- Can we achieve 2028x1520 / 4:3 / 50fps - Full FOV using 2x2 binning via the camera.txt file?

dgsiegel, 2020-11-26:
There's a pull request for the pi4: https://github.com/showmewebcam/showmewebcam/pull/13 

We've never tested this, as we only own a Pi Zero, but you're welcome to try and help us to get this merged.

AFAIK the max resolution is 1920x1080 at 30 fps due to hardware reasons. You might wanna try to ask this here, as this is the wrapper we use: https://github.com/peterbay/uvc-gadget

@htruong might be able to answer your question about wifi

htruong, 2020-11-27:
I have closed this issue but please let me know if you have any further questions.

closed 2020-11-27 by htruong

puccaso, 2020-11-28:
Thank you for your time and attention =]

You guys are great. I will be back with progress on the pi4 build

puc