showmewebcam/showmewebcam #47: Coil whining and strange artifacts

greimela, 2020-11-26:
Hi, thanks for this awesome project!
It was really easy to set it up on an Pi Zero W.

Is it normal that the Pi is whining as soon as the camera is active?

Also, after a couple of seconds I receive artifacts like these.
[image]

Any ideas what could be the reason for that or how to debug it?

dgsiegel, 2020-11-26:
I've never had any noise issues with my Pi. However I once noticed similar artifacts on a USB hub. Would you mind trying either a different cable or plugging the Pi directly into your computer?

greimela, 2020-11-26:
Thanks for the hint.

I tried it on a MacBook using an USB-C hub. And now I also tried it directly plugged into a Windows laptop. 

On both devices these artifacts appear after some seconds and the number of artifacts grows over time.

dgsiegel, 2020-11-26:
Would you mind reporting this to https://github.com/peterbay/uvc-gadget? It's the backend we use, and this looks like a hardware bug mostly

greimela, 2020-12-02:
Will do, thanks for your help!

closed 2020-12-02 by greimela