showmewebcam/showmewebcam #50: brightness setting in camera.txt is being ignored

zlorfi, 2020-11-27:
After writing `/boot/camera.txt` with
```
#Tweak the auto exposure bias
auto_exposure_bias=15
#Tweak the brightness
brightness=55
```
the `auto_expose_bias` value is being honoured, but not the `brightness` value. 
```
# /usr/bin/v4l2-ctl -L

User Controls

             brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=50 flags=slider
[....]          
             auto_exposure_bias 0x009a0913 (intmenu): min=0 max=24 default=12 value=15
                                0: -4000 (0xfffffffffffff060)
                                1: -3667 (0xfffffffffffff1ad)
                                2: -3333 (0xfffffffffffff2fb)
                                3: -3000 (0xfffffffffffff448)
                                4: -2667 (0xfffffffffffff595)
                                5: -2333 (0xfffffffffffff6e3)
                                6: -2000 (0xfffffffffffff830)
                                7: -1667 (0xfffffffffffff97d)
                                8: -1333 (0xfffffffffffffacb)
                                9: -1000 (0xfffffffffffffc18)
                                10: -667 (0xfffffffffffffd65)
                                11: -333 (0xfffffffffffffeb3)
                                12: 0 (0x0)
                                13: 333 (0x14d)
                                14: 667 (0x29b)
                                15: 1000 (0x3e8)
                                16: 1333 (0x535)
                                17: 1667 (0x683)
                                18: 2000 (0x7d0)
                                19: 2333 (0x91d)
                                20: 2667 (0xa6b)
                                21: 3000 (0xbb8)
                                22: 3333 (0xd05)
                                23: 3667 (0xe53)
                                24: 4000 (0xfa0)
[...]
```

Any idea how to fix this problem?

dgsiegel, 2020-11-27:
I can't seem to reproduce this. 

# v4l2-ctl -l
[...]
             brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=55 flags=slider
[...]
             auto_exposure_bias 0x009a0913 (intmenu): min=0 max=24 default=12 value=15

Could you paste your full `camera.txt` file, as well as the output of `journalctl`?

zlorfi, 2020-11-27:
```bash
# cat /boot/camera.txt 
#Tweak the auto exposure bias
auto_exposure_bias=15
#Tweak the brightness
brightness=55
```
below the output from `journalctl` (last lines which fit on my screen, I have no clue how to display the complete output)
```
Feb 07 15:50:57 piwebcam systemd[1]: Started Serial Getty on ttyAMA0.
Feb 07 15:50:58 piwebcam ifup[103]: ifup: can't open '/var/run/ifstate.new': No such file or directory
Feb 07 15:50:58 piwebcam systemd[1]: network.service: Main process exited, code=exited, status=1/FAILURE
Feb 07 15:50:58 piwebcam systemd[1]: network.service: Failed with result 'exit-code'.
Feb 07 15:50:58 piwebcam systemd[1]: Failed to start Network Connectivity.
Feb 07 15:50:58 piwebcam systemd[1]: Reached target Network.
Feb 07 15:50:58 piwebcam systemd[1]: Found device /dev/ttyGS0.
Feb 07 15:50:58 piwebcam systemd[1]: Started Serial Getty on ttyGS0.
Feb 07 15:50:58 piwebcam systemd[1]: Reached target Login Prompts.
Feb 07 15:50:58 piwebcam systemd[1]: Reached target Multi-User System.
Feb 07 15:50:58 piwebcam systemd[1]: Starting Update UTMP about System Runlevel Changes...
Feb 07 15:50:58 piwebcam systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Feb 07 15:50:58 piwebcam systemd[1]: Started Update UTMP about System Runlevel Changes.
Feb 07 15:50:58 piwebcam systemd[1]: Startup finished in 2.180s (kernel) + 6.930s (userspace) = 9.110s.
Feb 07 15:50:58 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_bind()
Feb 07 15:50:58 piwebcam kernel: dwc2 20980000.usb: bound driver configfs-gadget
Feb 07 15:50:59 piwebcam piwebcam[170]: Found camera.txt, applying settings
Feb 07 15:50:59 piwebcam piwebcam[182]: Setting auto_exposure_bias -> 15
Feb 07 15:50:59 piwebcam piwebcam[190]: Setting brightness -> 55
Feb 07 15:50:59 piwebcam kernel: dwc2 20980000.usb: new device is high-speed
Feb 07 15:50:59 piwebcam kernel: dwc2 20980000.usb: new address 18
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: high-speed config #2: c
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_set_alt(0, 0)
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: reset UVC Control
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_set_alt(1, 0)
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_set_alt(1, 0)
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_set_alt(1, 0)
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: Initial path: /sys/kernel/config/usb_gadget
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 10, resolution: 1600x1200, bitrate: [153600000, 307200000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 9, resolution: 1600x900, bitrate: [115200000, 230400000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 8, resolution: 1536x864, bitrate: [106168320, 212336640]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 7, resolution: 1920x1080, bitrate: [165888000, 331776000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 6, resolution: 1280x960, bitrate: [98304000, 196608000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 5, resolution: 1280x720, bitrate: [73728000, 147456000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 4, resolution: 1024x768, bitrate: [62914560, 125829120]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 3, resolution: 800x600, bitrate: [38400000, 76800000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 2, resolution: 640x480, bitrate: [24576000, 49152000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 1, resolution: 640x360, bitrate: [18432000, 36864000]
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: Number of buffers requested: 2
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: Video format: V4L2_PIX_FMT_MJPEG
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: Show FPS: DISABLED
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: IO method requested: USER_PTR
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: UVC device name: /dev/video1
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: V4L2 device name: /dev/video0
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_UVC: Opening /dev/video1 device
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_UVC: Device is 20980000.usb on bus gadget
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_V4L2: Opening /dev/video0 device
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_V4L2: Device is mmal service 16.1 on bus platform:bcm2835-v4l2
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_V4L2: Getting highest frame size: YUYV 4056x3040
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_V4L2: Getting highest frame size: MJPG 4056x3040
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Brightness (V4L2_CID_BRIGHTNESS = UVC_PU_BRIGHTNESS_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: 0, max: 100, step: 1, default: 50, value: 55
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 100, step: 1, default: 50, value: 55
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Contrast (V4L2_CID_CONTRAST = UVC_PU_CONTRAST_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: -100, max: 100, step: 1, default: 0, value: 0
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 200, step: 1, default: 100, value: 100
[... cut ...]
            image_stabilization 0x009a0916 (bool)   : default=0 value=0
                iso_sensitivity 0x009a0917 (intmenu): min=0 max=4 default=0 value=0
           iso_sensitivity_auto 0x009a0918 (menu)   : min=0 max=1 default=1 value=1
         exposure_metering_mode 0x009a0919 (menu)   : min=0 max=2 default=0 value=0
                     scene_mode 0x009a091a (menu)   : min=0 max=13 default=0 value=0

JPEG Compression Controls

            compression_quality 0x009d0903 (int)    : min=1 max=100 step=1 default=30 value=30
```

dgsiegel, 2020-11-27:
Hmm...
```
Feb 07 15:50:59 piwebcam piwebcam[182]: Setting auto_exposure_bias -> 15
Feb 07 15:50:59 piwebcam piwebcam[190]: Setting brightness -> 55
```
The settings are applied correctly. What happens if you issue 

`v4l2-ctl -c brightness=50`

manually?

zlorfi, 2020-11-27:
[image]

zlorfi, 2020-11-27:
> Hmm...
> 
> ```
> Feb 07 15:50:59 piwebcam piwebcam[182]: Setting auto_exposure_bias -> 15
> Feb 07 15:50:59 piwebcam piwebcam[190]: Setting brightness -> 55
> ```
> 
> The settings are applied correctly. What happens if you issue
> 
> `v4l2-ctl -c brightness=55`
> 
> manually?

Setting it manually works until the next reboot.
```
Your webcam at your service!
piwebcam login: root
Password: 
# v4l2-ctl -l

User Controls

                     brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=50 flags=slider
                       contrast 0x00980901 (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                     saturation 0x00980902 (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                    red_balance 0x0098090e (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                horizontal_flip 0x00980914 (bool)   : default=0 value=0
                  vertical_flip 0x00980915 (bool)   : default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=3 default=1 value=1
                      sharpness 0x0098091b (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                  color_effects 0x0098091f (menu)   : min=0 max=15 default=0 value=0
                         rotate 0x00980922 (int)    : min=0 max=360 step=90 default=0 value=0 flags=modify-layout
             color_effects_cbcr 0x0098092a (int)    : min=0 max=65535 step=1 default=32896 value=32896

Codec Controls

             video_bitrate_mode 0x009909ce (menu)   : min=0 max=1 default=0 value=0 flags=update
                  video_bitrate 0x009909cf (int)    : min=25000 max=25000000 step=25000 default=10000000 value=25000000
         repeat_sequence_header 0x009909e2 (bool)   : default=0 value=0
            h264_i_frame_period 0x00990a66 (int)    : min=0 max=2147483647 step=1 default=60 value=60
                     h264_level 0x00990a67 (menu)   : min=0 max=11 default=11 value=11
                   h264_profile 0x00990a6b (menu)   : min=0 max=4 default=4 value=4

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=0 value=0
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=1000 value=1000
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=0
             auto_exposure_bias 0x009a0913 (intmenu): min=0 max=24 default=12 value=15
      white_balance_auto_preset 0x009a0914 (menu)   : min=0 max=10 default=1 value=1
            image_stabilization 0x009a0916 (bool)   : default=0 value=0
                iso_sensitivity 0x009a0917 (intmenu): min=0 max=4 default=0 value=0
           iso_sensitivity_auto 0x009a0918 (menu)   : min=0 max=1 default=1 value=1
         exposure_metering_mode 0x009a0919 (menu)   : min=0 max=2 default=0 value=0
                     scene_mode 0x009a091a (menu)   : min=0 max=13 default=0 value=0

JPEG Compression Controls

            compression_quality 0x009d0903 (int)    : min=1 max=100 step=1 default=30 value=30
# v4l2-ctl -c brightness=55
# v4l2-ctl -l

User Controls

                     brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=55 flags=slider
                       contrast 0x00980901 (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                     saturation 0x00980902 (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                    red_balance 0x0098090e (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                horizontal_flip 0x00980914 (bool)   : default=0 value=0
                  vertical_flip 0x00980915 (bool)   : default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=3 default=1 value=1
                      sharpness 0x0098091b (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                  color_effects 0x0098091f (menu)   : min=0 max=15 default=0 value=0
                         rotate 0x00980922 (int)    : min=0 max=360 step=90 default=0 value=0 flags=modify-layout
             color_effects_cbcr 0x0098092a (int)    : min=0 max=65535 step=1 default=32896 value=32896

Codec Controls

             video_bitrate_mode 0x009909ce (menu)   : min=0 max=1 default=0 value=0 flags=update
                  video_bitrate 0x009909cf (int)    : min=25000 max=25000000 step=25000 default=10000000 value=25000000
         repeat_sequence_header 0x009909e2 (bool)   : default=0 value=0
            h264_i_frame_period 0x00990a66 (int)    : min=0 max=2147483647 step=1 default=60 value=60
                     h264_level 0x00990a67 (menu)   : min=0 max=11 default=11 value=11
                   h264_profile 0x00990a6b (menu)   : min=0 max=4 default=4 value=4

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=0 value=0
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=1000 value=1000
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=0
             auto_exposure_bias 0x009a0913 (intmenu): min=0 max=24 default=12 value=15
      white_balance_auto_preset 0x009a0914 (menu)   : min=0 max=10 default=1 value=1
            image_stabilization 0x009a0916 (bool)   : default=0 value=0
                iso_sensitivity 0x009a0917 (intmenu): min=0 max=4 default=0 value=0
           iso_sensitivity_auto 0x009a0918 (menu)   : min=0 max=1 default=1 value=1
         exposure_metering_mode 0x009a0919 (menu)   : min=0 max=2 default=0 value=0
                     scene_mode 0x009a091a (menu)   : min=0 max=13 default=0 value=0

JPEG Compression Controls

            compression_quality 0x009d0903 (int)    : min=1 max=100 step=1 default=30 value=30
# 
```

dgsiegel, 2020-11-27:
Could you try setting a value like 90 to see if is applied by uvc-gadget? Seems like it at least, based on the output:

```
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Brightness (V4L2_CID_BRIGHTNESS = UVC_PU_BRIGHTNESS_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: 0, max: 100, step: 1, default: 50, value: 55
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 100, step: 1, default: 50, value: 55
```

zlorfi, 2020-11-27:
setting the value for brightness to 90 in the `camera.txt file` has no effect, but I do get a different output from `journalctl `:
```
Feb 07 15:50:59 piwebcam piwebcam[170]: Found camera.txt, applying settings
Feb 07 15:50:59 piwebcam piwebcam[182]: Setting brightness -> 90
Feb 07 15:50:59 piwebcam piwebcam[190]: Setting auto_exposure_bias -> 15
Feb 07 15:50:59 piwebcam kernel: dwc2 20980000.usb: new device is high-speed
Feb 07 15:50:59 piwebcam kernel: dwc2 20980000.usb: new address 45
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: high-speed config #2: c
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_set_alt(0, 0)
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: reset UVC Control
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_set_alt(1, 0)
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_set_alt(1, 0)
Feb 07 15:50:59 piwebcam kernel: configfs-gadget gadget: uvc: uvc_function_set_alt(1, 0)
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: Initial path: /sys/kernel/config/usb_gadget
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 10, resolution: 1600x1200, bitrate: [153600000, 307200000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 9, resolution: 1600x900, bitrate: [115200000, 230400000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 8, resolution: 1536x864, bitrate: [106168320, 212336640]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 7, resolution: 1920x1080, bitrate: [165888000, 331776000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 6, resolution: 1280x960, bitrate: [98304000, 196608000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 5, resolution: 1280x720, bitrate: [73728000, 147456000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 4, resolution: 1024x768, bitrate: [62914560, 125829120]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 3, resolution: 800x600, bitrate: [38400000, 76800000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 2, resolution: 640x480, bitrate: [24576000, 49152000]
Feb 07 15:50:59 piwebcam piwebcam[192]: CONFIGFS: UVC: format: 1, frame: 1, resolution: 640x360, bitrate: [18432000, 36864000]
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: Number of buffers requested: 2
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: Video format: V4L2_PIX_FMT_MJPEG
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: Show FPS: DISABLED
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: IO method requested: USER_PTR
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: UVC device name: /dev/video1
Feb 07 15:50:59 piwebcam piwebcam[192]: SETTINGS: V4L2 device name: /dev/video0
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_UVC: Opening /dev/video1 device
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_UVC: Device is 20980000.usb on bus gadget
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_V4L2: Opening /dev/video0 device
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_V4L2: Device is mmal service 16.1 on bus platform:bcm2835-v4l2
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_V4L2: Getting highest frame size: YUYV 4056x3040
Feb 07 15:50:59 piwebcam piwebcam[192]: DEVICE_V4L2: Getting highest frame size: MJPG 4056x3040
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Brightness (V4L2_CID_BRIGHTNESS = UVC_PU_BRIGHTNESS_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: 0, max: 100, step: 1, default: 50, value: 90
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 100, step: 1, default: 50, value: 90
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Contrast (V4L2_CID_CONTRAST = UVC_PU_CONTRAST_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: -100, max: 100, step: 1, default: 0, value: 0
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 200, step: 1, default: 100, value: 100
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Saturation (V4L2_CID_SATURATION = UVC_PU_SATURATION_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: -100, max: 100, step: 1, default: 0, value: 0
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 200, step: 1, default: 100, value: 100
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Red Balance (V4L2_CID_RED_BALANCE + V4L2_CID_BLUE_BALANCE = UVC_PU_WHITE_BALANCE_COMPONENT_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: 1, max: 7999, step: 1, default: 1000, value: 1000
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 7998, step: 1, default: 999, value: 999
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Power Line Frequency (V4L2_CID_POWER_LINE_FREQUENCY = UVC_PU_POWER_LINE_FREQUENCY_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: 0, max: 3, step: 1, default: 1, value: 1
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 3, step: 1, default: 1, value: 1
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2: Supported control Sharpness (V4L2_CID_SHARPNESS = UVC_PU_SHARPNESS_CONTROL)
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   V4L2: min: -100, max: 100, step: 1, default: 0, value: 0
Feb 07 15:50:59 piwebcam piwebcam[192]: V4L2:   UVC: min: 0, max: 200, step: 1, default: 100, value: 100
[... cut ...]
Feb 07 15:51:04 piwebcam piwebcam[192]: DEVICE_UVC: 2 buffers allocated.
Feb 07 15:51:04 piwebcam piwebcam[192]: DEVICE_V4L2: STREAM ON success
Feb 07 15:51:04 piwebcam piwebcam[192]: DEVICE_UVC: STREAM ON success
Feb 07 15:51:04 piwebcam piwebcam[192]: DEVICE_V4L2: STREAM OFF success
Feb 07 15:51:04 piwebcam piwebcam[192]: DEVICE_UVC: STREAM OFF success
Feb 07 15:51:04 piwebcam piwebcam[192]: UVC: streaming request CS: PROBE, REQ: SET_CUR
Feb 07 15:51:15 piwebcam login[133]: root login on 'ttyGS0'
```

zlorfi, 2020-11-27:
Interesting side note, the vales seems to be reset:
```
Feb 07 15:51:04 piwebcam piwebcam[192]: UVC: PROCESSING_UNIT - GET_DEF - UVC_PU_BRIGHTNESS_CONTROL
Feb 07 15:51:04 piwebcam piwebcam[192]: UVC: PROCESSING_UNIT - GET_CUR - UVC_PU_BRIGHTNESS_CONTROL
Feb 07 15:51:04 piwebcam piwebcam[192]: UVC: PROCESSING_UNIT - SET_CUR - UVC_PU_BRIGHTNESS_CONTROL
Feb 07 15:51:04 piwebcam piwebcam[192]: UVC: control CONTROL_UNDEFINED, length: 2
Feb 07 15:51:04 piwebcam piwebcam[192]: DEVICE_V4L2: V4L2_CID_BRIGHTNESS changed value (V4L2: 50, UVC: 50)
```

dgsiegel, 2020-11-27:
Ok, this might seem to be related to https://github.com/peterbay/uvc-gadget

Would you mind reporting this bug there as well?

tuyenld, 2020-11-28:
> Interesting side note, the vales seems to be reset:
>
> ```
> Feb 07 15:51:04 piwebcam piwebcam[192]: UVC: PROCESSING_UNIT - GET_DEF - UVC_PU_BRIGHTNESS_CONTROL
> Feb 07 15:51:04 piwebcam piwebcam[192]: [...]

I doubt that your webcam application on the host side tried to change brightness value.
You can verify it by doing:
- wait til webcam boot.
- don't open any webcam application such as BBS, Zoom, Skype, etc on the host side (your computer).
- connect serial to Pi.
- verity brightness value: `v4l2-ctl -L | grep brightness`

peterbay, 2020-11-28:
answer is here - [https://github.com/peterbay/uvc-gadget/issues/11#issuecomment-735001499](https://github.com/peterbay/uvc-gadget/issues/11#issuecomment-735001499)

peterbay, 2020-11-28:
if you see a message "DEVICE_V4L2: V4L2_CID_BRIGHTNESS changed value (V4L2: 50, UVC: 50)" in your log, then the host computer sent a request for brightness change

mattstudio, 2020-12-01:
I'm encountering the same problem. Other settings in the camera.txt are being honored, but the 'brightness' value is ignored. I have to set it manually using the serial interface each time I connect the webcam.

Host computer is running macOS Catalina.

peterbay, 2020-12-02:
A possible solution for this problem is adding a switch for disabling controls from the host computer.
I will add this switch to the uvc-gadget.

dgsiegel, 2020-12-02:
Wouldn't it be better to patch V4L/ucv to expose all settings to the host computer? Guess not everybody wants to log in to the serial interface

mattstudio, 2020-12-02:
Having the setting persistent would really be amazing. I just want to say thank you to all the work you all have done to make this project as simple and plug-and-play as it is. It really does work reliably, boots quickly, and gets out of your way. While I was able to figure out how to use the serial interface to set this value, having to do it each time I connect my laptop to the webcam seems a bit of a chore. It's really cool to be able to adjust settings on the fly with over the serial interface, but I'd sacrifice that ability for a webcam that keeps my settings each time I connect it.

dgsiegel, 2020-12-02:
> having to do it each time I connect my laptop to the webcam seems a bit of a chore

Just put a `camera.txt` file with your settings in `/boot` :)

mattstudio, 2020-12-02:
> Just put a `camera.txt` file with your settings in `/boot` :)

That's the entire problem. I have added `brightness=55` to the `camera.txt` file on `/boot` but it gets ignored. Other settings are correctly applied but the `brightness` property doesn't work. I have to set it manually using the serial interface each time.

dgsiegel, 2020-12-02:
I am not saying that we shouldn't fix this bug. But you can use `auto_exposure_bias=20` as a workaround.

mattstudio, 2020-12-02:
Here's the difference I'm experiencing between using `auto_exposure_bias=20` and `brightness=55`
I see a very slight bump in brightness when I set `auto_exposure_bias=20` but setting `brightness=55` really makes a difference.
[image]

rossiniscarface, 2020-12-16:
> That's the entire problem. I have added `brightness=55` to the `camera.txt` file on `/boot` but it gets ignored. Other settings are correctly applied but the `brightness` property doesn't work. I [...]

Can't you change the brightness in your host-program? As i see, e.g. the Windows Camera App always initialize the cam with the last brightness i configured. So in the Camera App i change it to 55, in the boot/camera.txt it is 65 defined. If i open the app, the app changes the setting to 55. For a webcam i would want that the webcam does what the host commands and not what the webcam thinks it should do.

mattstudio, 2020-12-18:
@rossiniscarface I'm using macOS. I don't know of any OS-level way of adjusting a webcam's brightness setting and I haven't seen an app to control this setting. The OS very well might be sending a "set brightness to 50%" command, I just can't change it anywhere.

rossiniscarface, 2020-12-21:
hmmm osx it seems to doesn't have any control oob. I found some payed apps for camera-control but thats it. Very nice Apple. 
So from this standpoint it would be very good to have an option to set things in the camera and don't listen to the host-os. 
As a workaround you could write a little bash script which sends the brightness command to the showmewebcam.

jfurcean, 2020-12-21:
I use the Webcam Settings app on OSX - [http://mactaris.blogspot.com](http://mactaris.blogspot.com). This lets you set the brightness to the Piwebcam. Unfortunately, it doesn't look like any of the other settings are exposed (contrast, saturation, etc) like they are for my Logitech c920.

[image]

tuyenld, 2020-12-21:
@jfurcean 
Commit to enable more control setting #65 was merged. You will see it in the next release.
Or you can test a temporary image now : https://github.com/showmewebcam/showmewebcam/runs/1555578113 (click on `artifacts` near the top-right corner)

jfurcean, 2020-12-21:
> @jfurcean
> Commit to enable more control setting #65 was merged. You will see it in the next release.
> Or you can test a temporary image now : https://github.com/showmewebcam/showmewebcam/runs/1555578113 (click on `artifacts` near the top-right corner)

Awesome!

[image]

qpre, 2021-04-15:
> A possible solution for this problem is adding a switch for disabling controls from the host computer.
> I will add this switch to the uvc-gadget.

Hey ! stumbling upon the same issue were I want to disable the host machine from setting values, could you point me to where I could disable controls from the host computer ? thanks !

dgsiegel, 2021-08-22:
> Hey ! stumbling upon the same issue were I want to disable the host machine from setting values, could you point me to where I could disable controls from the host computer ? thanks !

That should be possible now: https://github.com/showmewebcam/showmewebcam#overriding-the-controls-available-to-the-host-computer

closed 2021-08-22 by dgsiegel