showmewebcam/showmewebcam #54: Seamless firmware upgrade from host

htruong, 2020-11-28:
One of the ways we can achieve seamless firmware upgrade is to have the following scheme:

- Enable networking-over-USB via the composite USB interface.
- Have a firmware upgrade daemon listening on a fixed port.
- Have two firmware partitions, A and B, one is active.
- At end of upgrade, switch the partitions in cmdline.txt.

peterbay, 2021-09-03:
And why don't make it simpler?
Preparation:
- create a binary file with filesystem FAT32 in firmware

Steps:
- connect to showmewebcam and initialize firmware upgrade
- mount prepared binary file and enable mass storage
- copy new firmware to USB mass storage from the host
- reboot
- mount binary file
- apply the stored file to a partition
- switch partitions in config
- reboot

Networking is not needed.

dgsiegel, 2021-09-06:
Unfortunately that won't work as the partition is read only. However I could easily imagine having two partitions on which you write the data via e.g. `dd`. For `/boot/` we would need to have some logic to see what's changed, but that's the easy part.