showmewebcam/showmewebcam #56: Proposed Text Based Interface Control App

AlexOD42, 2020-11-29:
Attached is a proposed lightweight app for adjusting the webcam settings in the terminal before committing to /boot/camera.txt. The only dependency should be libcurses5. At present, the v4l2 control schema is hard-coded in the app based on my ZeroW and HD Sensor. I haven't handled the /boot filesystem remount or perhaps there's a better way? There may be a few bugs hiding in there but thought it was worth sharing for others to test. Please feel free to add to your repo if it's useful.

Thanks for all the great work on the wider project!

Alex

[control.c.txt](https://github.com/showmewebcam/showmewebcam/files/5612643/control.c.txt)
[image]

htruong, 2020-11-29:
Thanks, let me add to new repo. I will credit you -- also do you want the source to be GPLv3 as well?

htruong, 2020-11-29:
The boot.txt remount is fine, it can be handled by a script, no big deal.

AlexOD42, 2020-11-29:
Perfect! Yes please to GPLv3.

dgsiegel, 2020-11-29:
Looks good! Agree that we should add a wrapper to handle camera.txt. Additionally, here's few thoughts:

- I think we should make use of a separate repo under the showmewebcam umbrella, as it would make it easier to integrate this tool via a buildroot package
- We would need a small Makefile, which handles compilation and installation of the binary (Here's an example you can use https://github.com/peterbay/uvc-gadget/blob/master/Makefile)
- Would it be possible to make the controls dynamic, ie. read them from v4l2? Here's some info on that: https://www.kernel.org/doc/html/v4.14/media/uapi/v4l/control.html
- You probably shouldn't use a hard coded filename in the code, make use of a parameter. That way you can also use it outside of showmewebcam
- Instead of using `system(cmd)` you want to use V4L2 controls like I mentioned above
- Do you already have a name for the app?

Love to include this for the next release, thanks for working on it!

peterbay, 2020-11-30:
I'm working on an implementation
[image]

peterbay, 2020-11-30:
aplha version of camera-control - [https://github.com/peterbay/camera-control](https://github.com/peterbay/camera-control)

dgsiegel, 2020-12-01:
Thanks @peterbay for stepping in! I've added a pr for this: #57 

Here's a few final ideas/topics we should discuss:

- Currently any existing comments as well as the order in camera.txt are overwritten if we save new settings. Is it worthwhile to keep them?
- Some settings won't work if others are not enabled, e.g. ISO Sensitivity, while White Balance was merged into one control
- Some settings are missing their actual names, e.g ISO Sensitivity could be 100, 200, 400, 800 instead of 1,2,3,4
- Should we filter unneeded controls, e.g. H264?

Other than that great job everybody!

peterbay, 2020-12-02:
Thanks @dgsiegel for the new ideas.
I've added a new version of camera-control.
Filtering of unneeded controls is now available through -i option. (like -i h264_i_frame_period)
Listing of all available controls is available through -l option. 
Added processing for controls of type V4L2 CTRL TYPE MENU - more controls have their correct options.  (ISO 100 ...)
At the moment, I don't know how to detect which controls will have no effect on the camera.

dgsiegel, 2020-12-02:
I've merged this. Thanks!

closed 2020-12-02 by dgsiegel