showmewebcam/showmewebcam #60: FPS Control

TheNewJavaman, 2020-12-03:
I'm using a Pi Zero (No W) with HQ Camera. Hypothetically, this device should support 720p60fps. How can I enable this setting?

htruong, 2020-12-12:
@TheNewJavaman Whenever possible can you help test the build in the PR to see if the framerate for 720p is actually 60fps? I see it listed here with this change, but I have no idea how to check if it's lying to me.

TheNewJavaman, 2020-12-15:
@htruong I'll try to get it done after this week. I have finals for the next few days.

htruong, 2020-12-16:
Good luck with your finals!

fgma, 2021-04-10:
@htruong You PR looks good to me. I've tested it using VLC on windows and v4l on linux. I even added 83333 to allow 120 FPS which would be great for machine vision applications like head tracking.

mspacek, 2021-06-04:
I just tested this (i.e. PR #72) by making the modifications myself on master. First I tried adding both 83333 and 166667 (10s of milliseconds? kind of a strange unit) at the same time for potentially 120 and 60 fps. That resulted in uvcvideo not working at all when plugging it in to my linux laptop. My pi0w + HQ only showed up as a USB device. Tried again with just 166667, uvcvideo works, and 60 fps shows up as an option in guvcview (http://guvcview.sourceforge.net/), but it never actually delivers more than 30 fps. Same goes for vlc. So I guess it's more complicated than this simple edit. Perhaps it needs to be resolution specific (#128)?

htruong, 2021-06-04:
@mspacek Just curious, how do you verify whether the video is actually 60fps or it's faux 60fps?

mspacek, 2021-06-04:
I can instantly tell just looking at the onscreen preview. Everything is much more alive at 60 fps. Try a youtube video and compare normal to 60 fps, you should be able to see it too. Also, guvcview displays the fps in the window title of the preview window, and vlc displays fps in the Tools->Codec information dialog.

Al3601, 2021-11-10:
any news on this? i tested the PR, we can select 720p60fps but it is actually 30fps, it is driver related ? im actually runing an IR OV5647  with Pi0W.
as i can run 720p 60fps with this camera with the latest Pios, im sure there is something forgotten here.

HunterAP23, 2022-02-03:
I also tested this patch and can confirm that regardless of resolution or using MJPEG/uncompressed, the camera will only output a max of 30fps even when set to 60fps mode. Recording just the webcam in OBS at 60fps with no reported encoding/rendering lag, then manually checking individual frames, the recording duplicates the previous frame once - effectively running it at 30fps instead of the intended 60fps.

I noticed that the related PR #72 was closed but not merged, since apparently just adding `166666` to `package/piwebcam/multi-gadget.sh`'s the list under the `/dwFrameInterval` definition does not work.

I couldn't find any branches / forks attempting to add this in, and I'm not skilled enough to figure it out myself - but if people need testers then I'll certainly help out