showmewebcam/showmewebcam #63: Cannot find camera-ctl

msurguy, 2020-12-07:
It appears that  /usr/bin/camera-ctl doesn't exist on the RPI 0 image currently.

I flashed it a couple days ago using https://github.com/showmewebcam/showmewebcam/releases/download/v1.60/sdcard-raspberrypi0-v1.60.img.gz

Wondering where that file is?

dgsiegel, 2020-12-07:
That's is probably my fault 🙈️ `camera-ctl` is part of the current repository, but did not make it into the last release. It will be part of 1.70 though.

You can either build showmewebcam yourself or try one of our development snapshots, e.g. https://github.com/showmewebcam/showmewebcam/actions/runs/404447718

closed 2020-12-07 by dgsiegel

htruong, 2020-12-07:
@dgsiegel should we create a "stable" branch and set it as default, so it shows the README of the stable one?

dgsiegel, 2020-12-08:
We should rather create another release :) I think this was my mistake as this information should be part of the wiki and not the README

htruong, 2020-12-08:
Sure, let's do a release!