showmewebcam/showmewebcam #67: Webcam no longer shows up

9j7axvsLuF, 2020-12-10:
Hi, thanks for your great work on this repo.

It was working perfectly for me, but today after a reboot the webcam doesn't show up anymore. I've tried reflashing the firmware (both versions), but it doesn't change anything. The microSD card seems to be working correctly.

In case that's helpful: when I plug the raspberri pi zero, I see the green LED flashing a few times, then nothing.

Thanks in advance for your help!

mattstudio, 2020-12-10:
Check the cable that connects the pi to the camera. Has it come loose at all?

9j7axvsLuF, 2020-12-10:
Hi Matt,

Thanks - I already checked both of these things. Also it's unlikely to be a connection issue since it happened after a reboot (a forced reboot actually, after a Zoom call froze my laptop).

Any idea what it might be? I doubt the RPi Zero board is defective, it's lighting up at least upon booting on the microSD card.

htruong, 2020-12-11:
I have the issue the other day too. Most likely the cable is bad.

You can check the hardware by flashing a functional raspberry pi os to it, and then use the utility to try capturing the image.

9j7axvsLuF, 2020-12-16:
Thanks. I checked the hardware, the pi is fine. And to my surprise, at some point the camera started working again. However, it stopped booting again today. I think the cable is fine, I wonder what it can be.

htruong, 2020-12-20:
@9j7axvsLuF  Try installing my build: https://github.com/showmewebcam/showmewebcam/pull/79

Before you plug camera in do `$ sudo dmesg -w` and watch it.

If you see that only the `ttyACM` device show up, then your camera or camera cable is bad.

If you see nothing show up at all, raspberry pi, usb port, or usb cable is bad.

htruong, 2020-12-27:
Let's try with this image actually: https://github.com/htruong/showmewebcam/actions/runs/447449067

A-Ghamgosar, 2021-01-31:
Hi. thanks for your repo. how can I solve this problem:
I use an external power source to power up the Rpi. I want to plug and unplug the camera without waiting for bootup time. when the camera is in use by software on my laptop and I unplug it, the camera no longer shows up and I have to reboot the pi to make it work. but when the camera is not in use, I can plug and replug it as many times as I want and my laptop recognizes it well.

AblfzlGh, 2021-02-01:
> Hi. thanks for your repo. how can I solve this problem:
> I use an external power source to power up the Rpi. I want to plug and unplug the camera without waiting for bootup time. when the camera is [...]

update: I could solve this by killing the process and running it again. 
killall /opt/uvc-webcam/uvc-gadget
. /opt/uvc-webcam/start-webcam.sh /bin/sh /opt/uvc-web &

It can be done by reading gpio21 state but I don't know how to do it yet :))

closed 2021-04-20 by dgsiegel