showmewebcam/showmewebcam #68: /usr/bin/camera-ctl missing in sdcard-raspberrypi0w-v1.60.img.gz

nokolade, 2020-12-11:
Hi all,
First time user, impressively convenient solution!
Following README.md, I found /usr/bin/camera-ctl to be missing in sdcard-raspberrypi0w-v1.60.img.gz

"evidence":
# find / -name *camera*
/boot/camera.txt
/run/udev/data/+platform:bcm2835-camera
/sys/kernel/config/usb_gadget/piwebcam/functions/uvc.usb0/control/terminal/camera
/sys/devices/platform/soc/2000b840.mailbox/bcm2835-camera
/sys/bus/platform/devices/bcm2835-camera
/sys/bus/platform/drivers/bcm2835-camera
/sys/bus/platform/drivers/bcm2835-camera/bcm2835-camera
/sys/module/bcm2835_v4l2/drivers/platform:bcm2835-camera
/usr/lib/modules/5.4.73/kernel/drivers/leds/trigger/ledtrig-camera.ko
/usr/lib/modules/5.4.73/kernel/drivers/staging/vc04_services/bcm2835-camera

rossiniscarface, 2020-12-11:
Duplicate of #63 which is marked closed, but there is no new release.

dgsiegel, 2020-12-11:
Right, it was only available when you built showmewebcam yourself. Until now: https://github.com/showmewebcam/showmewebcam/releases/tag/v1.70

closed 2020-12-11 by dgsiegel