showmewebcam/showmewebcam #74: How to improve Image Quality

rossiniscarface, 2020-12-12:
Is it possible to improve the image quality? 
I think the mjpg compression is very bad at the highest setting. The lowest setting is absolute not usable. Is it possible to change the range of the compression setting. 
Sometimes low framerate and high image quality is better than high framerate and low image quality.
Is it possible to use uncompressed yuv mode?

dgsiegel, 2020-12-12:
It is possible to use uncompressed yuv, however you won't be happy with that either, as we only have about 7-8fps.

The quality is mostly based on the `video_bitrate`. Did you set it to max?

rossiniscarface, 2020-12-12:
`video_bitrate=25000000 ` is the default and the max.

dgsiegel, 2020-12-12:
You would have to add a new configfs configuration, ie. editing `multi-gadget.sh`. Instead of `config_frame mjpeg m $width $height` just use `config_frame uncompressed u $width $height`.

We had enabled that a few months ago but disabled it due to the poor performance. Also make sure you have the latest uvc-gadget.

Lastly, I think raspistill/raspivid is able to get a slightly better quality than we are able via V4L2. Would be great if you could share any findings.

htruong, 2020-12-12:
So like if someone really wants to keep a still frame maybe we should provide raspistill to do pictures. That might be better than to figure out how to take a still frame from the video which is limited to 1080p anyways.

rossiniscarface, 2020-12-13:
Looked at the code:
First finding: there are two v4l2 drivers: bcm2835-v4l2 and uv4l-raspicam. 
So some things just don't work, because the driver doesn't implement that. E.g the compression_quality does nothing for pixel format mjpg. The compression quality for mjpg is only handled by the video-bitrate which is capped hard coded by 25*1000*1000 in bcm2835-camera/controls.c. Why? Didn't find out. Maybe setting this cap higher could improve quality. 

The bcm2835-v4l2 module is initialized with MAX_VIDEO_MODE_WIDTH 1280 and MAX_VIDEO_MODE_HEIGHT 720. Don't think thats an issue. 
The Framerate is hardcoded capped at #define FPS_MAX 90 ... 

Changing the Framerate with v4l2-ctl -p on anything higher than 50 on the resolution of 1920x1080 the video quality gets very pixelated but fast ... change it to 50 in a 4:3 mode the raspberry hangs. (Putty console and stream)

Overall the findings i had in the internet on the bcm2835-v4l2 driver in combination with mjpg are not that good. Looks like better results are with uv4l-raspicam in combination with mjpg or using h264/h265.

dgsiegel, 2020-12-13:
> First finding: there are two v4l2 drivers: bcm2835-v4l2 and uv4l-raspicam.

uv4l is the userspace V4L2 module. From what I've seen so far,  `bcm2835-v4l2` should be more advanced than the userspace module.

> the compression_quality does nothing for pixel format mjpg.

Correct, compression_quality is only for stills, not video.

> The compression quality for mjpg is only handled by the video-bitrate which is capped hard coded by 25_1000_1000 in bcm2835-camera/controls.c. Why? 

I guess this is a hardware limit: 

> Maximum bitrate is 25Mbits/s (-b 25000000), but much over 17Mbits/s won't show noticeable improvement at 1080p30.
https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

> 
> The bcm2835-v4l2 module is initialized with MAX_VIDEO_MODE_WIDTH 1280 and MAX_VIDEO_MODE_HEIGHT 720. Don't think thats an issue.

That doesn't seem to apply to the V4L2 module from what I understand.
 
> Overall the findings i had in the internet on the bcm2835-v4l2 driver in combination with mjpg are not that good. Looks like better results are with uv4l-raspicam in combination with mjpg or using h264/h265.

H264 is not implemented, see also https://github.com/peterbay/uvc-gadget/issues/8

Were you able to test uv4l on your Raspberry? Seems like it supports h264 and if it delivers better performance, I have no issues at all switching to that.

rossiniscarface, 2020-12-13:
> Maximum bitrate is 25Mbits/s (-b 25000000), but much over 17Mbits/s won't show noticeable improvement at 1080p30.

this is true for h264 but not for mjpg as far i understand. 

> That doesn't seem to apply to the V4L2 module from what I understand.
> 
> >

But its hardcoded in the driver and you can look them up in the sys/modules/.... parameter?

> H264 is not implemented, see also

Yeah H264 is not implemented in the uvc-gadget modul. But the bcm2835-v4l2 module already support it. 

Switching the camera driver to uv4l wont change the fact that uvc-gadget doesn‘t support h264 (yet?)

dgsiegel, 2020-12-13:
> But its hardcoded in the driver and you can look them up in the sys/modules/.... parameter?

`v4l2-ctl -L` works as well.

> Switching the camera driver to uv4l wont change the fact that uvc-gadget doesn‘t support h264 (yet?)

Sure, but that we could fix :)

dgsiegel, 2020-12-15:
> The compression quality for mjpg is only handled by the video-bitrate which is capped hard coded by 25_1000_1000 in bcm2835-camera/controls.c. Why? Didn't find out. Maybe setting this cap higher could improve quality.

You were right, I could increase this to at least 35_1000_1000 by patching the v4l2 module.

rossiniscarface, 2020-12-15:
Nice. Did you see a change in picture quality from 25000000 to 350000000
Do you have a image to test?

dgsiegel, 2020-12-15:
Barely visible. And a lag appears starting with 60000000.

I'd rather investigate how to reduce analog/digital gain. That seems to reduce the quality much more.

rossiniscarface, 2020-12-16:
> Barely visible. And a lag appears starting with 60000000.
> 
> I'd rather investigate how to reduce analog/digital gain. That seems to reduce the quality much more.
> 
> Attached is an image where you can try it yourself: [sdcard.zip](https://github.com/showmewebcam/showmewebcam/files/5699287/sdcard.zip)

Hmm, image doesn't boot on my zero w. Is it for the zero or zero w?

dgsiegel, 2020-12-16:
It's for the zero. But you can easily build an image yourself, as I only changed a single line in the kernel module.

Noguai, 2020-12-16:
> I'd rather investigate how to reduce analog/digital gain. That seems to reduce the quality much more.

Is that what the ``auto_exposure_bias`` setting primarily influences? 

Because I too found image quality to be lacking at first, but mostly because of massive massive amounts of noise, which mostly disappeared after reducing ``auto_exposure_bias`` from it's default value ``12`` down to ``2`` and countered the loss in brightness by setting

```
brightness=68
contrast=42
sharpness=33
video_bitrate=25000000
auto_exposure_bias=2
exposure_metering_mode=2
```

This gives me an image with much less noise, and seemingly better dynamic range. I can still see some blocking in darker areas of the image, but it's fine for me now.

I thought about creating an issue about the default settings, but disregarded that idea, because I guess everybody will have to and will tune the settings anyway to their own lighting conditions. Though I have an idea about the behavior of the brightness slider which kills contrast at the moment...

dgsiegel, 2020-12-16:
> Is that what the `auto_exposure_bias` setting primarily influences?

Yes and no. From my understanding, the camera has an analog and digital gain setting, which is not writeable via V4L2 but only through MMAL right now. Exposure mode, Exposure bias, ISO and Scene mode contribute to the actual value of the gain settings. More is explained here: https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-gain

I am still exploring how these settings relate to each other. Any help or feedback would be very welcome.

towolf, 2021-01-05:
If you lack light you can enable `night` mode.

dogadogan, 2022-01-10:
Any updates on manually setting analog and digital gain settings? @dgsiegel