showmewebcam/showmewebcam #77: boot without SD card and use rpiboot instead

Janghou, 2020-12-17:
This is a great project.

Maybe most use-cases are using an SD card and plugging the Pi in, I can see a use case for booting the piwebcam WITHOUT an SD-card but using usbboot/rpiboot, just like the RPI gpioexpander tool.

https://github.com/raspberrypi/usbboot
https://github.com/raspberrypi/usbbootgui

That will save the cost for an SD card, which is the most expensive part in a budget setup. More expensive then a Zero, or clone V1 camera.

Not sure how difficult that is.

My first thought: pack the root dir as an initramfs.cpio.gz and served it from the boot directory (initramfs rootfs.cpio.gz followkernel), but I have no real experience with it.

htruong, 2020-12-20:
It's not that difficult. However, then we're depending on the host OS to start the webcam. This is very hairy since we will have to write a separate application for each target OS we want to support to deliver the OS payload to the Pi Zero. Combined with the amount of architectures we have to support (and potential quirks of each system), you can see how quickly this can become a nightmare. We have at least 3 different OSes, each OS has at least two architectures (x86 and ARM).

You (or anyone else reading this) are welcome to create an app to do that, but I don't think that anyone in the team is going to undertake this effort.

closed 2020-12-20 by htruong

Janghou, 2020-12-20:
My idea is of course to use the existing `usbboot/rpiboot` application.
`rpiboot` works by booting a Pi in device mode from directories containing the booting firmware etc.

And starting the Pi by the guest OS offers the possibility to easily adjust settings in the camera.txt before boot, no need for mounting and editing.

But it's a different use-case, so I have forked the project. 
https://github.com/Janghou/showmewebcam-usbboot

As I'm totally new to buildroot, I'm not sure where to start. 

I think the first thing to try is to integrate the rootfs in the kernel image if that's possible. Idea and help  is welcome.

dgsiegel, 2020-12-20:
> And starting the Pi by the OS offers the possibility to easily adjust settings in the camera.txt before boot, no need for mounting and editing.

You can connect through the serial console and use camera-ctl. Or help us find better ways to allow adjusting the settings.

htruong, 2020-12-20:
> As I'm totally new to buildroot, I'm not sure where to start.

There is the all in one manual at https://buildroot.org/downloads/manual/manual.html that I still refer to from time to time.

I have started some documentation effort [here but not much has been done](https://github.com/showmewebcam/showmewebcam/wiki/Development-guide).

We document most interesting things we see online here: https://github.com/showmewebcam/showmewebcam/wiki/Further-Documentation

Janghou, 2020-12-21:
@dgsiegel, I will when I find one.

@htruong, most examples of buildroot work with creating an SD card image with two partitions. I don't need that. Major challenge is to provide a rootfs without having the second partition on the SD card (there is none), therefor I'm trying to find a way to embed the rootfs in the kernel. 

Or use NFS boot. Something like described here, booting a [RPI Zero with Raspbian without SD card](https://dev.webonomic.nl/how-to-run-or-boot-raspbian-on-a-raspberry-pi-zero-without-an-sd-card/comment-page-1). That is slow though.

Janghou, 2021-01-08:
FYI I have a working fork booting without SD-card:
https://github.com/Janghou/showmewebcam-usbboot