showmewebcam/showmewebcam #80: kernel uvcvideo errors on master: Failed to query (GET_INFO) UVC control

antoneliasson, 2020-12-20:
https://github.com/showmewebcam/showmewebcam/commit/8ba054b25ef8a5c2993473088795deac4588c5e4 introduced kernel errors that are printed when the device is plugged in and queried:

> dec 20 11:51:54 miramis kernel: usb 1-1.3: new full-speed USB device number 11 using ehci-pci
> dec 20 11:51:58 miramis kernel: usb 1-1.3: new high-speed USB device number 12 using ehci-pci
> dec 20 11:51:58 miramis kernel: usb 1-1.3: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
> dec 20 11:51:58 miramis kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> dec 20 11:51:58 miramis kernel: usb 1-1.3: Product: Piwebcam 
> dec 20 11:51:58 miramis kernel: usb 1-1.3: Manufacturer: Show-me Webcam Project
> dec 20 11:51:58 miramis kernel: usb 1-1.3: SerialNumber: 100000000d2386db
> dec 20 11:51:58 miramis kernel: uvcvideo: Found UVC 1.00 device Piwebcam  (1d6b:0104)
> dec 20 11:51:58 miramis kernel: uvcvideo: **Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).**
> dec 20 11:51:58 miramis kernel: uvcvideo: **Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).**
> dec 20 11:51:58 miramis kernel: uvcvideo: **Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).**
> dec 20 11:51:58 miramis kernel: uvcvideo: **Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).**
> dec 20 11:51:58 miramis kernel: input: Piwebcam : UVC Camera as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:2.0/input/input31
> dec 20 11:51:58 miramis kernel: cdc_acm 1-1.3:2.2: ttyACM0: USB ACM device

The lines marked in bold are printk's at KERN_ERR level. They were not shown when plugging in a device with firmware built from tag v1.7.0.

I'm using a RPi 0 and my host system is Ubuntu 20.04 (kernel 5.4).

antoneliasson, 2020-12-23:
Comparing kernel logs (journalctl -t kernel) on the RPi between v1.7.0 and a93e0318 shows a few new prints at info level:

~~~
Sep 12 12:16:48 piwebcam kernel: dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep (ptrval) ep0, 1)
Sep 12 12:16:48 piwebcam kernel: dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep (ptrval) ep0, 1)
Sep 12 12:16:48 piwebcam kernel: dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep (ptrval) ep0, 1)
Sep 12 12:16:48 piwebcam kernel: dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep (ptrval) ep0, 1)
~~~

A similar print is shown every time I start a stream with VLC (which causes a `uvcvideo: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).` on the host).

tuyenld, 2021-03-27:
Duplicate #116

closed 2021-04-20 by dgsiegel