showmewebcam/showmewebcam #86: Manual ISO changes do not take effect, but can be triggered by flipping image

larsjuhljensen, 2020-12-29:
Using the new **camera-ctl** tool (on release 1.70) with **ISO Sensitivity, Auto** set to **Manual**, changes to **ISO Sensitivity** do not take effect. However, doing a **Horizontal Flip** of the image (and back again) is sufficient to make the ISO change take effect.

I am running the 1.70 release on a Raspberry Pi Zero with the HQ camera.

dgsiegel, 2021-01-01:
Hmm.. I can't really reproduce this here. What are your initial settings?

larsjuhljensen, 2021-01-01:
It does not seem to matter with other settings for me. However, it is easier to see if you turn **Auto Exposure** off and set an **Exposure Time, Absolute**. With fixed exposure time, it is immediately obvious when the new ISO takes effect, due to the change in image brightness.

dgsiegel, 2021-01-04:
Oh, now I got you: You need to set the preferred ISO, *then* doing a horizontal flip to see any changes, right?

larsjuhljensen, 2021-01-04:
Precisely! Whenever setting a new ISO value, it has no effect until I do a flip.

dgsiegel, 2021-01-04:
Alright, I think we need to move this upstream in this case. Would you mind reporting this to https://github.com/raspberrypi/linux ?

Additionally we would be happy to add some kind of workaround until we get this fixed. Open for any ideas or suggestions.

larsjuhljensen, 2021-01-04:
I am not sure what to report to the upstream here, since I do know what exactly your code does.

Regarding workaround, I think it could be done pretty easily. I have just noticed that if auto_exposure is set to 0, ISO changes appear to take effect immediately. So in camera-ctl (or whichever tool it calls downstream), when a user wants to change iso_sensitivity, simply set auto_exposure to 0, then apply the iso_sensitivity change, and set auto_exposure back to its original value. That should be all it takes to work around this bug.

dgsiegel, 2021-01-04:
> I am not sure what to report to the upstream here, since I do know what exactly your code does.

Exactly what you reported here :) We're basically just using the kernel's v4l2 module which uses the Raspberry's MMAL interface.

> Regarding workaround, I think it could be done pretty easily. I have just noticed that if auto_exposure is set to 0, ISO changes appear to take effect immediately. So in camera-ctl (or whichever [...]

Noted, that should be easy enough to add to camera-ctl.

larsjuhljensen, 2021-01-04:
It looks like the issue is precisely what has been reported in 2016 and closed without being solved. Not sure what to do about that:
https://github.com/raspberrypi/linux/issues/1440

It was pointed out in 2018 that, despite it being closed, this is a real issue. This sounds identical to what I describe:
https://github.com/raspberrypi/linux/issues/1440#issuecomment-375492384

dgsiegel, 2021-01-04:
It seems that they think that this is not really what's going on here: https://github.com/raspberrypi/linux/issues/1440#issuecomment-302363481

larsjuhljensen, 2021-01-04:
And us reporting again, what others have already reported will likely not change anything. The comment from dstarke describes exactly what I see as well, but it seems like nobody even responded to him.

dgsiegel, 2021-01-04:
Sure, but he might also be right. In other words: It could be that EV, analog/digital gain and other automations will be triggered when flipping the image, but the ISO change was effective in all cases?

larsjuhljensen, 2021-01-04:
I'm running with auto_exposure turned off and fixed exposure_time_absolute precisely to do manual exposure and not have any automation get it wrong. Flipping the image or changing auto_exposure on and back off triggered the ISO change in all cases for me.

kbingham, 2021-01-04:
Open a new issue on https://github.com/raspberrypi/linux and reference the closed one, with a description that is fully reproducible.

The response on the closed issue probably got ignored ... because it was already closed.

biasedlogic, 2021-01-19:
I have submitted a new bug report https://github.com/raspberrypi/linux/issues/4081

dgsiegel, 2021-01-27:
Until this gets resolved, we might think about some workarounds, e.g. flipping the webcam every time after we change some settings. I've discovered that it also sometimes resets the settings, when you use the webcam for a new stream.

dgsiegel, 2021-02-24:
From further testing it seems that settings get reset, but can be re-applied with flipping the image. Any patches are welcome.