showmewebcam/showmewebcam #87: Corrupted video stream followed by crash (Pi Zero W)

ChickenSupreme, 2021-01-01:
I've been running a previous version of uvc-gadget (this one: https://github.com/climberhunt/uvc-gadget) on my Pi Zero W for a few months now without problems. I decided to upgrade to the latest and greatest, particularly since some apps don't support 1080p webcams so it would be nice to have support for multiple resolutions.

I wrote the image to the SD card with `sudo dd if=~/Downloads/sdcard-raspberrypi0w-v1.70.img of=/dev/sdb` and booted it up. It came up quickly and flashed the LED three times so I tested it with ffplay:

```
$ ffplay /dev/video0
ffplay version 4.3.1 Copyright (c) 2003-2020 the FFmpeg developers
  built with gcc 10 (GCC)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --docdir=/usr/share/doc/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=x86_64 --optflags='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection' --extra-ldflags='-Wl,-z,relro -Wl,--as-needed -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld ' --extra-cflags=' ' --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libvo-amrwbenc --enable-version3 --enable-bzlib --disable-crystalhd --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-libaom --enable-libdav1d --enable-libass --enable-libbluray --enable-libcdio --enable-libdrm --enable-libjack --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-liblensfun --enable-libmp3lame --enable-libmysofa --enable-nvenc --enable-openal --enable-opencl --enable-opengl --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librav1e --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libvorbis --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-version3 --enable-vapoursynth --enable-libvpx --enable-vulkan --enable-libglslang --enable-libx264 --enable-libx265 --enable-libxvid --enable-libzimg --enable-libzvbi --enable-avfilter --enable-avresample --enable-libmodplug --enable-postproc --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --disable-stripping --shlibdir=/usr/lib64 --enable-lto --enable-libmfx --enable-runtime-cpudetect
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
[mjpeg @ 0x7f12180022c0] Found EOI before any SOF, ignoring f=0/0   
[mjpeg @ 0x7f12180022c0] No JPEG data found in image
Input #0, video4linux2,v4l2, from '/dev/video0':B sq=    0B f=0/0   
  Duration: N/A, start: 5378.438004, bitrate: N/A
    Stream #0:0: Video: mjpeg (Baseline), yuvj420p(pc, bt470bg/unknown/unknown), 640x360, 30 fps, 30 tbr, 1000k tbn, 1000k tbc
[mjpeg @ 0x7f121800ab40] Found EOI before any SOF, ignoring
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[swscaler @ 0x7f121406f2c0] deprecated pixel format used, make sure you did set range correctly
[mjpeg @ 0x7f121800ab40] Found EOI before any SOF, ignoring f=0/0   
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] overread 80KB vq=    0KB sq=    0B f=0/0   
[mjpeg @ 0x7f121800ab40] EOI missing, emulating
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] Found EOI before any SOF, ignoring
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] overread 80KB vq=    0KB sq=    0B f=0/0   
[mjpeg @ 0x7f121800ab40] EOI missing, emulating
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] Found EOI before any SOF, ignoring f=0/0   
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] Found EOI before any SOF, ignoring f=0/0   
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] overread 80KB vq=    0KB sq=    0B f=0/0   
[mjpeg @ 0x7f121800ab40] EOI missing, emulating
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] Found EOI before any SOF, ignoring
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] overread 80KB vq=    0KB sq=    0B f=0/0   
[mjpeg @ 0x7f121800ab40] EOI missing, emulating
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] Found EOI before any SOF, ignoring
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
[mjpeg @ 0x7f121800ab40] overread 80KB vq=    0KB sq=    0B f=0/0   
[mjpeg @ 0x7f121800ab40] EOI missing, emulating
[mjpeg @ 0x7f121800ab40] No JPEG data found in image
```

The above errors are repeated hundreds of times, accompanied by visible corruption in the video stream (blocks of incorrect or mangled pixels). I tested it a few times and always after 5-10 seconds or so the video stream freezes and the Pi's LED goes out. I also tried with mpv and got the same result.

One test ended with these messages:

```
[mjpeg @ 0x7fa11800ab40] Found EOI before any SOF, ignoring
[mjpeg @ 0x7fa11800ab40] No JPEG data found in image
[mjpeg @ 0x7fa11800ab40] No JPEG data found in image=    0B f=0/0   
[mjpeg @ 0x7fa11800ab40] No JPEG data found in image=    0B f=0/0   
[mjpeg @ 0x7fa11800ab40] Found EOI before any SOF, ignoring
[mjpeg @ 0x7fa11800ab40] No JPEG data found in image
[mjpeg @ 0x7fa11800ab40] overread 80KB vq=    0KB sq=    0B f=0/0   
[mjpeg @ 0x7fa11800ab40] EOI missing, emulating
[mjpeg @ 0x7fa11800ab40] No JPEG data found in image
[video4linux2,v4l2 @ 0x7fa118000bc0] Dequeued v4l2 buffer contains corrupted data (0 bytes).
[video4linux2,v4l2 @ 0x7fa118000bc0] ioctl(VIDIOC_DQBUF): No such device
ioctl(VIDIOC_QBUF): No such device
[video4linux2,v4l2 @ 0x7fa118000bc0] ioctl(VIDIOC_DQBUF): No such device
    Last message repeated 2 times
[video4linux2,v4l2 @ 0x7fa118000bc0] ioctl(VIDIOC_DQBUF): No such device
    Last message repeated 2 times
[video4linux2,v4l2 @ 0x7fa118000bc0] ioctl(VIDIOC_DQBUF): No such device
    Last message repeated 2 times
```

Here's my dmesg output:

```
[ 5358.630000] usb 1-1: new full-speed USB device number 6 using xhci_hcd
[ 5362.602915] usb 1-1: new high-speed USB device number 7 using xhci_hcd
[ 5362.832210] usb 1-1: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[ 5362.832212] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5362.832213] usb 1-1: Product: Piwebcam 
[ 5362.832214] usb 1-1: Manufacturer: Show-me Webcam Project
[ 5362.832214] usb 1-1: SerialNumber: 100000000d2386db
[ 5362.862087] uvcvideo: Found UVC 1.00 device Piwebcam  (1d6b:0104)
[ 5362.886241] input: Piwebcam : UVC Camera as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-1/1-1:2.0/input/input31
[ 5362.895120] cdc_acm 1-1:2.2: ttyACM0: USB ACM device
[ 5439.898716] usb 1-1: USB disconnect, device number 7
[ 5439.902500] uvcvideo: Failed to resubmit video URB (-19).
[ 5439.906500] uvcvideo: Failed to resubmit video URB (-19).
[ 5439.910500] uvcvideo: Failed to resubmit video URB (-19).
[ 5439.914500] uvcvideo: Failed to resubmit video URB (-19).
[ 5439.918499] uvcvideo: Failed to resubmit video URB (-19).
```

ChickenSupreme, 2021-01-01:
I just tried the 1.6.0 and 1.5.0 releases, and a different USB cable, with no change.

htruong, 2021-01-01:
You should try with another csi flex cable. Or if you don't have a spare, try with stock raspbian and stream video from the camera there to see if you have the same issue...

ChickenSupreme, 2021-01-01:
Up until yesterday I was using the same pi/camera/sdcard/cable/etc as a webcam with uvc-gadget by following this guide: http://www.davidhunt.ie/raspberry-pi-zero-with-pi-camera-as-usb-webcam/

I also tested with a spare USB cable and had the same issue. Both USB cables worked fine previously.

ChickenSupreme, 2021-01-01:
OK so I flashed my SD card with stock Raspbian Lite and built and installed peterbay's uvc-gadget like so:

```
sudo apt install git
git clone https://github.com/peterbay/uvc-gadget.git
sudo cp ~/uvc-gadget/piwebcam.service /etc/systemd/system/
sudo systemctl enable piwebcam
# added "modules-load=dwc2,libcomposite" to /boot/cmdline.txt
# added "dtoverlay=dwc2" to /boot/config.txt
cd ~/uvc-gadget
make
sudo ln -s /lib/systemd/system/getty@.service /etc/systemd/system/getty.target.wants/getty@ttyGS0.service
```

I booted it up and it's working properly so I'm not sure why it's not working with showmewebcam. I will probably just leave it running as is but I do have a spare SD card and would be happy to do further tests if you'd like me to. I'm just not sure what to try next myself.

htruong, 2021-01-03:
One hunch I have is [this change](
https://github.com/showmewebcam/showmewebcam/pull/35/files#diff-eeb71b8eb45c8b327eb3bb3ddcf58c10ade29fd0d995e5bd32e8d17cbdfe8735R66)

Could you help removing that `initial_turbo` line from `config.txt` and see if the situation improves for you?

ChickenSupreme, 2021-01-03:
Yes that fixed it, thank you!

Keep up the great work on this project by the way. I think it's super important to have open hardware/firmware and I love how fast it boots and not having to worry about cutting the power anymore.

closed 2021-01-03 by ChickenSupreme

dgsiegel, 2021-01-03:
I've added this to the FAQ, thanks!

htruong, 2021-01-03:
I wonder because it mysteriously breaks for someone, should we do this as a opt-in instead of opt-out?

I'm in favor of making it work for everybody.

dgsiegel, 2021-01-03:
Thought about the same thing. I'd say if more people appear with this or similar issues, we should remove it/make it optional. Until then I'd keep it as it seems to work for most people so far.