showmewebcam/showmewebcam #88: Cannot compile piwebcam: it includes header files of the host kernel

oskar456, 2021-01-03:
Hello,

Im trying to build this for the first time with 2020.02.9 on my Gentoo system (amd64 running vanilla kernel 5.9.9). The compilation goes on for a while but then it fails during building the `piwebcam` package:

```
>>> piwebcam 28605a4f79fe6bb9de9a3e2babfa0ce22c668e5f Building
/usr/bin/make -j9 CXX="/home/oskar/Downloads/showmewebcam/output/raspberrypi0/host/bin/arm-buildroot-linux-gnueabihf-g++" CC="/home/oskar/Downloads/showmewebcam/output/raspberrypi0/host/bin/arm-buildroot-linux-gnueabihf-gcc" LD="/home/oskar/Downloads/showmewebcam/output/raspberrypi0/host/bin/arm-buildroot-linux-gnueabihf-ld" -C /home/oskar/Downloads/showmewebcam/output/raspberrypi0/build/piwebcam-28605a4f79fe6bb9de9a3e2babfa0ce22c668e5f
/home/oskar/Downloads/showmewebcam/output/raspberrypi0/host/bin/arm-buildroot-linux-gnueabihf-gcc -W -Wall -g -I/usr/src/linux/include -I/usr/src/linux/arch/x86/include   -c -o uvc-gadget.o uvc-gadget.c
In file included from /usr/src/linux/arch/x86/include/uapi/asm/sigcontext.h:18,
                 from /usr/src/linux/arch/x86/include/asm/sigcontext.h:7,
                 from /home/oskar/Downloads/showmewebcam/output/raspberrypi0/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/bits/sigcontext.h:30,
                 from /home/oskar/Downloads/showmewebcam/output/raspberrypi0/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/signal.h:291,
                 from uvc-gadget.c:50:
/usr/src/linux/include/linux/compiler.h:232:10: fatal error: asm/rwonce.h: No such file or directory
 #include <asm/rwonce.h>
          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [<builtin>: uvc-gadget.o] Error 1
make[1]: *** [package/pkg-generic.mk:269: /home/oskar/Downloads/showmewebcam/output/raspberrypi0/build/piwebcam-28605a4f79fe6bb9de9a3e2babfa0ce22c668e5f/.stamp_built] Error 2
make: *** [Makefile:23: _all] Error 2
```

From the Makefile in the `output/raspberrypi0/build/piwebcam-28605a4f79fe6bb9de9a3e2babfa0ce22c668e5f`, it is obvious that it is actually trying to include host kernel header files from `/usr/src/linux` and architecture `x86`. I'm not sure how to fix this since I haven't find similar header files anywhere inside the buildroot output directory.

Any help would be appreciated.

oskar456, 2021-01-03:
Actually, it turned out that this can be fixed by simply not including anything at all. `uvc-gadget` will then build without issues.

dgsiegel, 2021-01-03:
Seems like your kernel headers don't have those header files installed. In any case, our build script should not use the system's header files, but the ones in the build directory. So this is definitely something we should fix.

Additionally I don't have the linux headers installed at all at this location, so I am wondering if we actually need them.
Would you mind putting the attached patch into `package/piwebcam/` (no need to apply, just save it there with the `.patch` extension) and rebuild?

You might need the remove the piwebcam build directory first (`output/raspberrypi0/build/piwebcam-*`).

[0001-piwebcam-remove-kernel-headers.patch.txt](https://github.com/showmewebcam/showmewebcam/files/5762344/0001-piwebcam-remove-kernel-headers.patch.txt)

htruong, 2021-01-04:
We should submit this patch to the `uvc-gadget` upstream.

dgsiegel, 2021-01-04:
> We should submit this patch to the `uvc-gadget` upstream.

If it works for everybody :) I've only tried it locally

oskar456, 2021-01-04:
> Additionally I don't have the linux headers installed at all at this location, so I am wondering if we actually need them.
It seems that including the header files from kernel sources will actually confuse the compiler. Most people probably don't have kernel sources at `/usr/src/linux` so nothing gets included and everything works.

> Would you mind putting the attached patch into `package/piwebcam/` (no need to apply, just save it there with the `.patch` extension) and rebuild?
> 
Thank you! I can confirm that this resolves my issue.

dgsiegel, 2021-01-04:
Alright, thanks for testing! I'll try to get this merged upstream. Thanks for reporting!

closed 2021-01-04 by dgsiegel