showmewebcam/showmewebcam #90: Resolution with Teams on Windows 10 is too low

westfeld, 2021-01-12:
Hello everyone,

I am experiencing an issue when using the webcam on a Windows 10 PC with Teams. The resolution is not set to 1080p when using the camera. I checked this using the serial interface and camera-ctl.

**After startup:**
Resolution: 1024x768
Format: JPEG

**When using Teams (preview camera in Teams' settings and in an actual video meeting)**
Resolution: 640x360
Format: MJPG

Right now my workaround is to use OBS and the virtual camera to loop the video through it and indeed I get a nice 1080p HD output from the camera. It would be easier to just do the right thing and not having to fiddle around with OBS.

rdcuk, 2021-02-19:
I have the same experience.

Am unable to use the Windows Camera app trick mentioned elsewhere. Really need the ability to override the resolution which teams wants. Could it be added to config somewhere?

I know we're effectively making up for the lack of settings in Teams here, but I guess we're all stuck with it :-(

dgsiegel, 2021-02-24:
We currently set up all resolutions in `package/piwebcam/multi-gadget.sh`. It would be easy to outsource those into a config file in /boot and make them configurable there. Feel free to send a PR.

rdcuk, 2021-02-24:
Sounds good. I wouldn't really know where to start on this myself (unless you want it re-impementing in COBOL), but if anyone has the time/inclination then I would be happy to test.

westfeld, 2021-04-11:
I have now rebuilt a new variant of Showmewebcam and just left out the following resolutions in `package/piwebcam/multi-gadget.sh`

    -  config_frame mjpeg m  640  360
    -  config_frame mjpeg m  640  480
    -  config_frame mjpeg m  800  600
    -  config_frame mjpeg m 1024  768

At first sight it looks better, but I will test that. Once it has been confirmed that it works better, we can think about making it configurable more easily.

westfeld, 2021-04-13:
After more testing with Teams on Windows 10, it works flawlessly in 720p resolution.

rdcuk, 2021-04-15:
Is there any way to change this on the standard SD card? i.e. without building a specific image on another system

dgsiegel, 2021-04-15:
No, but we'd be happy to accept a patch making this configurable, e.g. with a file in /boot.

closed 2021-08-22 by dgsiegel