showmewebcam/showmewebcam #91: Not recognized out of the box as webcam on Windows 10

KillaMaaki, 2021-01-13:
Figured I'd open this as a quality of life issue, but on my Windows 10 box the pi cam was not initially treated as a webcam. In the device manager it showed up as a "Linux USB Ethernet/RNDIS Gadget" but with an initialize error. I managed to fix it by choosing "Update drivers" and forcing Windows to use USB camera drivers (and have confirmed the camera working now!), but it would've been nice to not need this step.

Without knowing a whole lot, I suspect an issue in the USB descriptor data. I used a tool to capture a dump of the USB descriptor data, but I don't know quite enough about device descriptors to know what might be causing Windows to not treat it as a webcam. I've attached the dump from the utility, perhaps somebody else can take a look?

[DescriptorDump_Piwebcam_.txt](https://github.com/showmewebcam/showmewebcam/files/5810927/DescriptorDump_Piwebcam_.txt)

htruong, 2021-01-13:
I bet you tested the PR with the networking enable patch. This is an unfortunate side effect of being generous to test PRs :) Thank you, I don't think it will happen to a normal user though.

Windows, unfortunately, has the bad habit of "remembering" what devices was plugged to it, and cache what it knows about the device.

GlaireDaggers, 2021-01-13:
I just grabbed whatever the latest release image was, v1.70. Did that contain an experimental PR? Was under the impression that those were stable releases.

htruong, 2021-01-13:
That's interesting. I have a pretty fresh version of Windows and didn't face the issue you had. I will have a second look to the issue and see if I can reproduce the issue.

Have you ever, by chance, installed/tested on the Pi something that make the Pi emulate the USB gadget and plugged into the computer before?

GlaireDaggers, 2021-01-13:
This is my first Pi, a fresh Pi Zero W, and the only thing I have ever flashed to its SD was the showmewebcam image.

htruong, 2021-01-14:
Thanks for the info. I will look into the issue and let you know what I found out.

Lo5t, 2021-02-22:
+1, new Pi Zero W with 1.7 release

Rolrayn, 2021-02-24:
+1, Pi Zero w/o Wifi with 1.7 release

Rolrayn, 2021-02-24:
> +1, Pi Zero w/o Wifi with 1.7 release

I couldn't get version 1.6 or 1.5 to work either...

closed 2021-08-22 by dgsiegel