showmewebcam/showmewebcam #95: Raspi Zero W 1.1 not booting

mikeseeh, 2021-01-27:
Using the sdcard-raspberrypi0w-v1.70.img image a Raspi Zero W 1.1 is not booting.
After the rainbow square it prints the following 2 lines and is then stuck:
```
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```
Official Raspberry Pi images are booting fine.

dgsiegel, 2021-01-27:
We had several similar reports. Can you try to either use a different cable or to remove `initial_turbo` like explained here: https://github.com/showmewebcam/showmewebcam/wiki/FAQ#my-raspberry-pi-doesnt-work ?

mikeseeh, 2021-01-28:
Tested now …
- 2 different Pi Zero W
- 3 USB cables
- different USB ports (direct and hub)
- removed initial_turbo

Unfortunately it's now working.

kraftmac, 2021-01-29:
Did you make sure you have a Raspi Zero W? I was having booting issues and realized I had accidentally ordered the regular Zero and was using the sdcard-raspberrypi0w-v1.70.img. Mine works with appropriate image!

quincunx, 2021-01-29:
Having the same problem as @mikeseeh, tried with two Raspi Zero W 1.1, two different SD cards and two different cables.

So I went for older releases - release v1.40 (sdcard-raspberry0w-v1.40.img.gz) works (device got detected as USB camera and I could view the camera stream). Starting with release v1.50 it doesn't boot anymore (monitor only shows the above mentioned two lines).

dgsiegel, 2021-01-29:
We have introduced some boot time improvements in 1.50 which changed some kernel settings and might be the culprit here. However there are many people who have successfully booted 1.70 with their Zero W....

htruong, 2021-01-29:
Yeah, let's try to revert the speed boost, I think it might have hurted more than it helped.

dgsiegel, 2021-01-29:
> Yeah, let's try to consider reverting the speed boost tweak, I think it might have hurted more than it helped. I wonder how many seconds does it save.

I'd rather find out the exact issue, as it works for more people than not

quincunx, 2021-01-29:
I tried to compile the image and it takes ages on my computer, but I could test different builds to pinpoint the causing change.

htruong, 2021-01-30:
You can remove the `initial_turbo` directly on the sd card, no need to recompile.

quincunx, 2021-01-30:
@htruong I did try with removed `initial_turbo` line, too (forgot to mention that).

But - I think I've found the cause for the problem and can confirm that the latest build (v1.70) works. I'm afraid that the camera module might not have been connected properly - which causes the USB webcam device not to show up.

@mikeseeh Can you check if the camera module is connected properly, please? Remove the `quiet` in `cmdline.txt` to see the full kernel output.

sphere300, 2021-02-01:
Either manually building with the latest version (3f61042e2f41565c64e25f44860c4f9e4fb63cbf), or usingthe latest tagged 1.70 release I can still reproduce this issue. using 1.4 does work for me as well.

For myself noticed that 1.7 worked for a while but was a bit flakey on startup (IE I had to unplug and restart it several times before it would work) and recently has stopped working at all with the same error as reported elsewhere: 

brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

I wonder if the reason not more people are reporting the issue is that they are just restarting until it works. I would be happy to test the issue here as well if you need another test point. I only have 1 Raspberrypi 0w though.

dgsiegel, 2021-02-01:
The problem is that the above lines are not really warnings or errors, they are part of the normal kernel output. Would you mind trying to check the camera cable as well? We've changed our systemd scripts and they should fail if the camera was not present at boot.

sphere300, 2021-02-02:
Sorry had some work stuff to sort out. Yes the camera works with the normal desktop raspistill.
I just retested it using two different SD Cards, 1 up to date Raspbian image, and 1x that was setup one at a time with 1.7, master, and 1.4 on it. 

I tested Raspbian and could take pictures and video using the camera.

I tested 1.7, then master and 1.4 and 1.7 and master show the timeout still, and 1.4 loads but never seems to load in control panel. 

Afterwards just to be safe I retested Raspbian to ensure that yes it still can take a picture. 

I was using 1.7 since mid December till last weekend with intermittent start up issues. If you would like I can open a different issue ticket, but the behavior matches the initial description to a T. Unfortunately aside from the timeout there is not much 1.7 outputs that would help illuminate what's going wrong.

sphere300, 2021-02-04:
Ok I retested once more with 1.7 and can confirm that after testing a bunch of USB cables that I can get some to work. The best I can figure is that the power connections are working but either the data is not a solid connection or the port on the PI 0W USB port is slowly wearing out. Its interesting that this is cryptic when it fails, but it seems that I did not test thoroughly enough and will order a slew of new data cables to replace the old ones. Sorry to take your time on this, and thanks for your help.

kekscode, 2021-02-05:
I see the same issues as @mikeseeh mentioned above and did try the same things to mitigate. The only thing i could not test yet is another data cable between camera and pi zero w 1.1 . I ordered 3 different ones in order to test that, too. I also don't see the tty on Mac and the PI's green LED blinks 3 times and then it is off.

I was able to install motioneye os just fine, but the camera did not work OOTB. I'll install raspbian for testing the camera and the cable now.

jsiddall, 2021-03-18:
Update on this: I had the exact opposite problem.  I had setup a Zero successfully and recently got a Zero W.  I tried to write the same image, not realizing there was a separate image for the zero W, and had the issue of getting hung at the color screen.  I did check the FAQ and found nothing so it might be a good idea to put an entry there: if your Pi gets stuck at the color screen then double check that your Pi model matches the image version.

closed 2021-04-20 by dgsiegel

pazz, 2021-05-25:
Hi! I'm also running into this problem and am stuck after trying everything mentioned above.

I have a pi zero w 1.1 and the hq camera attached and I'm using `sdcard-raspberrypi0w-v1.80.img.gz` from the releases page.

The pie boots and I can log into the serial console. I've commented out the `initial_turbo` line in `boot/config.txt` and removed the ` quiet` suffix in `boot/cmdline.txt`.

Not having used this hardware for anything else so far, to me the only problematic line in the output is regarding the firmware (brcm/brcmfmac43430-sdio.bin). [Full output here] (https://gist.githubusercontent.com/pazz/bcfafc2eeb77a4a5a2d42d91f732ff8b/raw/35d33c69f219d695af7e9a682d33cf360fb64a80/gistfile1.txt) the interesting bit is below.

```
usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.10
usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
usb usb1: Product: DWC OTG Controller
usb usb1: Manufacturer: Linux 5.10.11 dwc2_hsotg
usb usb1: SerialNumber: 20980000.usb
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 1 port detected
vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
mc: Linux media interface: v0.10
bcm2835_vc_sm_cma_probe: Videocore shared memory driver
videodev: Linux video capture interface: v2.00
[vc_sm_connected_init]: start
[vc_sm_connected_init]: installed successfully
vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
cfg80211: Loading compiled-in X.509 certificates for regulatory database
bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
cfg80211: failed to load regulatory.db
brcmfmac: F1 signature read @0x18000000=0x1541a9a6
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
usbcore: registered new interface driver brcmfmac
brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2
bcm2835-codec bcm2835-codec: Device registered as /dev/video10
bcm2835-codec bcm2835-codec: Loaded V4L2 decode
bcm2835-codec bcm2835-codec: Device registered as /dev/video11
bcm2835-codec bcm2835-codec: Loaded V4L2 encode
bcm2835-codec bcm2835-codec: Device registered as /dev/video12
brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
bcm2835-codec bcm2835-codec: Loaded V4L2 isp
dwc2 20980000.usb: bound driver configfs-gadget
dwc2 20980000.usb: new device is high-speed
dwc2 20980000.usb: new device is high-speed
uart-pl011 20201000.serial: no DMA platform data
dwc2 20980000.usb: new device is high-speed
dwc2 20980000.usb: new address 22
```

I think I attached the video module correctly but as can be seen from the log above, there is no `/dev/video0` registered, only 10,11,12.

```
# ls /dev/video*
/dev/video10  /dev/video11  /dev/video12
```

Finally, this is what I get on the USB host on the other end of the cable:

```
 [127890.068039] usb 1-4: new full-speed USB device number 24 using xhci_hcd
[127894.979888] usb 1-4: new high-speed USB device number 25 using xhci_hcd
[127895.141593] usb 1-4: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[127895.141602] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[127895.141608] usb 1-4: Product: Piwebcam
[127895.141611] usb 1-4: Manufacturer: Show-me Webcam Project
[127895.141615] usb 1-4: SerialNumber: 100000000d2386db
[127895.145803] cdc_acm 1-4:2.0: ttyACM0: USB ACM device
```

I have tried several USB cables. 
Thanks for any insights!

pazz, 2021-05-25:
It seems that the camera cable was not correctly attached. Sorry for the noise.

dgsiegel, 2021-05-25:
It's always the camera cable :) Will add that to the FAQ. Thanks!