showmewebcam/showmewebcam #96: Camera output grayed out

atticus112, 2021-02-01:
UVC camera and USB serial devices show up, serial interfacing works, but camera output is grayed out in OBS and Zoom crashes when the UVC camera is selected. I am using a third party camera module - could this be a compatibility issue?

atticus112, 2021-02-01:
Using a Pi Zero W with 1.70

akseidel, 2021-02-17:
Perhaps. I am using the same Pi Zero W with 1.70. The camera shows up just fine as "Piwebcam: UVC camera" in Webcamoid running on a Pi 4b. I don't have OBS or Zoom loaded on that computer but I do on some Macs. OBS and Zoom on those computers see the Piwebcam. I don't believe the "UVC camera" is part of the video source name when using the Macs.

atticus112, 2021-02-22:
Turns out my camera module was broken. Works great with the replacement NoIR module I picked up!

closed 2021-02-22 by atticus112