showmewebcam/showmewebcam #98: The file is read-only

AblfzlGh, 2021-02-03:
I need to test some ideas and I want to edit or add some files and services in /etc/systemd/... but I can't.
how can I:
1. add/edit/remove some files in the directions except /boot ???
2. If it is a hardware lock, what should I change in commands before building a new version from your code and buildroot?

tuyenld, 2021-02-04:
@AblfzlGh 
Currently, there is no command build to do that. However, you can do it manually. Here is an instruction:
1. Revert this change https://github.com/showmewebcam/showmewebcam/pull/82/commits/6b67c18285573e1ef8f4be11f7a65a58eaf6b0c5
2. Change Pi0/Pi0w board configuration on https://github.com/showmewebcam/showmewebcam/blob/master/board/genimage-raspberrypi0.cfg
```
 image = "rootfs.squashfs" to  image = "rootfs.ext2" or  "rootfs.ext4"
```

tuyenld, 2021-02-05:
I forgot to mention you still need to change `rootfstype ` in `/boot/cmdline.txt`
```
rootfstype=squashfs -> rootfstype=ext2 or ext4
```

Rworld-Ravi, 2021-02-05:
@tuyenld 
Hi , I did rebuild the image after reverting the command set in config file and cfg file, also made changes in cmdline.txt , but now the pi does not boot as a webcam and nothing happens. any suggestion??

Rworld-Ravi, 2021-02-05:
Hey , ext4 worked for me.

closed 2021-02-24 by dgsiegel