showmewebcam/showmewebcam #99: Issue with Raspberry Pi 0 w/o wifi

blazczak, 2021-02-04:
After following the installation steps, the usb gadget is not returning images. 5MP image sensor

Built with buildroot-2020.02.9:

```
BUILDROOT_DIR=~/build/buildroot-2020.02.9 ./build-showmewebcam.sh raspberrypi0
```

```
$ lsusb
Bus 001 Device 012: ID 1d6b:0104 Linux Foundation Multifunction Composite Gadget
```

When the Pi0 is attached to Ubuntu, here's the dmesg output:
```
[ 4584.005208] usb 1-1.5.4: new high-speed USB device number 11 using dwc_otg
[ 4584.393219] usb 1-1.5.4: device descriptor read/64, error -71
[ 4587.605259] usb 1-1.5.4: device descriptor read/64, error -71
[ 4587.893256] usb 1-1.5.4: new high-speed USB device number 12 using dwc_otg
[ 4588.094977] usb 1-1.5.4: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[ 4588.095003] usb 1-1.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4588.095022] usb 1-1.5.4: Product: Piwebcam
[ 4588.095040] usb 1-1.5.4: Manufacturer: Show-me Webcam Project
[ 4588.095058] usb 1-1.5.4: SerialNumber: 100000000dxxxxdb
[ 4588.097808] uvcvideo: Found UVC 1.00 device Piwebcam (1d6b:0104)
[ 4588.102131] uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[ 4588.111439] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
[ 4588.114764] uvcvideo: Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).
[ 4588.119585] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
[ 4588.130254] input: Piwebcam: UVC Camera as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.4/1-1.5.4:2.0/input/input3
```

Trying to capture a video frame results in a pitch black image:
```
fswebcam -d /dev/video0 -v --jpeg 95 -r 2592x1944 -save test.jpg -F 1
--- Opening /dev/video0...
Trying source module v4l2...
/dev/video0 opened.
src_v4l2_get_capability,87: /dev/video0 information:
src_v4l2_get_capability,88: cap.driver: "uvcvideo"
src_v4l2_get_capability,89: cap.card: "Piwebcam: UVC Camera"
src_v4l2_get_capability,90: cap.bus_info: "usb-3f980000.usb-1.5.4"
src_v4l2_get_capability,91: cap.capabilities=0x84A00001
src_v4l2_get_capability,92: - VIDEO_CAPTURE
src_v4l2_get_capability,103: - STREAMING
No input was specified, using the first.
src_v4l2_set_input,181: /dev/video0: Input 0 information:
src_v4l2_set_input,182: name = "Camera 1"
src_v4l2_set_input,183: type = 00000002
src_v4l2_set_input,185: - CAMERA
src_v4l2_set_input,186: audioset = 00000000
src_v4l2_set_input,187: tuner = 00000000
src_v4l2_set_input,188: status = 00000000
src_v4l2_set_pix_format,520: Device offers the following V4L2 pixel formats:
src_v4l2_set_pix_format,533: 0: [0x47504A4D] 'MJPG' (Motion-JPEG)
Using palette MJPEG
Adjusting resolution from 2592x1944 to 1920x1080.
src_v4l2_set_mmap,672: mmap information:
src_v4l2_set_mmap,673: frames=4
src_v4l2_set_mmap,722: 0 length=2073600
src_v4l2_set_mmap,722: 1 length=2073600
src_v4l2_set_mmap,722: 2 length=2073600
src_v4l2_set_mmap,722: 3 length=2073600
--- Capturing frame...
verify_jpeg_dht,94: Inserting DHT segment into JPEG frame.
GD Warning: gd-jpeg: JPEG library reports unrecoverable error: Not a JPEG file: starts with 0x0b 0x8aCaptured frame in 0.00 seconds.
--- Processing captured image...
Setting output format to JPEG, quality 95
Writing JPEG image to 'test.jpg'.
```

Any ideas?
There was an irq-related "[Firmware bug]" message in the Pi0's console output with latest master (I'm assuming 1.80) so I downgraded to v1.70, but still no joy.

AblfzlGh, 2021-02-06:
Did you try using IMG versions attached [here](uhttps://github.com/showmewebcam/showmewebcam/tagsrl) and writing them to your sd card?

blazczak, 2021-02-08:
Yes, I used commit 2a039b1 to build what I reference above as v1.70, wrote the resulting sdcard.img file to the SD card and got the output above. I'm not interested in the prebuilt images available in "Releases".
Is there something amiss in the custom build, described in section "Development & building"?

dgsiegel, 2021-04-20:
Would you mind trying the release images? Another thing you might want to check are the cables, one for the camera and the USB cable. Most issues people reported so far were because of those.

closed 2021-08-22 by dgsiegel