showmewebcam/showmewebcam #103: Requesting an LED heartbeat when video is not being transmitted

akseidel, 2021-02-13:
After the initial showmewebcam boot sequence that terminates with the three Pi Zero LED flashes, showmewebcam illuminates the Pi Zero LED when video is being USB transmitted. That signal confirmation is great. 

Depending on how the system is being used there can be times when video is not being requested by the connected computer. During those times there is no external evidence the showmecam system is functioning in the ready state. A double LED flash every four seconds or so would be helpful.

dgsiegel, 2021-02-13:
Why do you need evidence? If it is plugged in, it should be working. At least that is our goal :)

But I think it would be interesting to have it the other way round, e.g. flashing when an error occurred or it's not working due to some reason?

akseidel, 2021-02-13:
I beg your pardon, but there are showmewebcam use situations that you may not be aware of. Its usefulness and who uses it goes beyond that of the computer savvy technologist. There are use instances where the system would be running but not actively feeding video. There are also other reasons for "not working" that is not the result of showmewebcam. I have personally seen USB cables to blame that reliably worked on one computer system but not at all on another. Then again there is always the video source not properly selected. The average user would not know to wait for the three LED flashes for confirmation. I myself have missed it even when I thought I looked soon enough. The heartbeat flash would help during troubleshooting and it also would provide assurance the system is running and ready to feed video. 

I was also going to suggest an error condition flash, but I thought that would be a secondary concern, especially since I have no idea what goes on inside to consider what of kind of running error the system could detect.