showmewebcam/showmewebcam #104: Low resolution with Camera module V2

EmadMokhtar, 2021-02-14:
I installed the OS on my Raspberry Pi Zero W with Raspberry Camera module V2. The resolution II got when I open the camera settings is **640x360**. I know the camera module V2 is capable of more than that. How to change the resolution?

EmadMokhtar, 2021-02-14:
I managed to change the resolution using this command.

``` bash
v4l2-ctl --set-fmt-video=width=1640,height=1232,pixelformat=H264
```

closed 2021-02-14 by EmadMokhtar

rdcuk, 2021-02-21:
Hi,
I'm having the same issue regarding resolution also (from Teams & Skype, I've tried your command but it seems to get reset when the camera is used. When the camera is in use then it won't allow the setting to be changed. Did you have the same issue?
Thanks,
Robin.