showmewebcam/showmewebcam #110: Video stream and when hardware triggered, take image and send via USB

neherh, 2021-02-25:
Hello, 

I am wondering if there is a possibility to stream the video and when a GPIO pin triggers, to take an image and send the image via the USB? How one might develop this?

Chimrod, 2021-05-23:
Yes, I'm using something similar (but not user friendly…)

I'm using a custom image with: 
- [v4l2grab](https://github.com/twam/v4l2grab) (for the capture)
- [lrzsz](https://ohse.de/uwe/software/lrzsz.html) (file transfert over serial connection)

On the PC, you will need a terminal which allow file transfert ([picocom](https://github.com/npat-efault/picocom), [minicom](https://salsa.debian.org/minicom-team/minicom)…). If you are using linux, all the tools can be installed throught the package manager. On windows this a bit more complicated (I'm using cygwin, putty is not able to use the file transfert. picocom build out of source directly).

Now connect with the serial interface and take a picture (like describe in #140). Launch the file transfert with:

    sz /tmp/picture.jpg

and press `^A^R` (in picocom) in order to receive the file. Press enter on the prompt, and voilà ! You got the picture on you PC.

I didn't found a way to compile `rz` on windows, and fallback to the [Xmodem](https://en.wikipedia.org/wiki/XMODEM) protocol. In order to transfert the file you run:

    sx -k /tmp/picture.jpg

in attachment, you will get a python script which retrieve the file (as rx would do). Run picocom with thoses arguments to activate it 

    picocom/./picocom.exe --receive-cmd "python rx.py" -b 115200 /dev/…

I didn't say it is easy, but there is a solution !

file:///tmp/rx.py

neherh, 2021-05-24:
Thanks for a solution. I am honestly suprised that V4L doesn't have an image taking capability. Otherwise, this would just be easy.

Chimrod, 2021-05-24:
Of course you could ! But your are limited by the image resolution available for streaming ! Which is less lower than the max resolution available : 

Example on my desktop :

    $ ./v4l2grab -W 3280 -H 2464 -q 100 -o /tmp/picture.jpg
    Image width set to 1920 by device /dev/video0.
    Image height set to 1080 by device /dev/video0.

It's a pity when you have the HQ camera plugged to your raspberry !