showmewebcam/showmewebcam #111: Can´t change settings

Lippsson, 2021-02-26:
Hello!
Thanks for the great work and code. Its was really simple to make it happend with my raspberz zero w and hq sensor. Now i want to change some setting (for example test how 1080p with 30fps work) and find some instructions in the Wiki ->Networking (On Windows)
I am on the point where i add the 2 file (Disable-webcam) and (enable-usb-network) the /boot partition, but it shows the webcam anyway. 

[image]

Can you please help me out here?
Thank you

htruong, 2021-04-04:
I'm yet to merge the patch that enables networking, so stay tuned :)

htruong, 2021-04-04:
https://github.com/showmewebcam/showmewebcam/pulls

closed 2021-04-04 by htruong