showmewebcam/showmewebcam #112: Unable to change resolution from default

HunterAP23, 2021-03-03:
Using a Raspberry Pi Zero W with a 4GB SD card.
Everything works fine, except that the camera resolution defaulted to 640x480 and I am unable to change it.
Normally you can change it with `v4l2-ctl --set-fmt-video=width=1280,height=720` but because the camera device is in-use by the script, I get the error message: `VIDIOC_S_FMT: failed: Device or resource busy`.

My initial idea of killing the process that is running `start-webcam.sh` would then let me set the resolution and then I could simply run the script manually, but killing the process causes the Raspberry Pi to shut down.

My 2nd guess would've been to edit the `start-webcam.sh` script to run that above command before the `uvc-gadget` command in that script is run, but as the file system is read only I can't change it.

I've looked through the documentation and issues but there doesn't seem to be any mention of it outside of closed issue #104 which was closed without any solution.

rcarmo, 2021-03-13:
I'd also like an easy way to switch resolutions here. Trying to free up the uvc device almost inevitably reboots the camera, which is a nuisance.

dberzon, 2021-03-16:
I also can't change the resolution, so the solution would be very helpful.
Thanks a lot for a great job!

HunterAP23, 2021-03-22:
My suggestions would be something along these lines:
1. Add some command or option in `/usr/bin/camera-ctl` to change the resolution. Changing the resolution stops the UVC camera app/service, changes the resolution, and the starts the service.
2. Make it so when running the new command or `/usr/bin/camera-ctl` change the resolution, the Pi stops the camera service but does not reboot, BUT the Pi sill reboots if the service/app is stopped otherwise.

closed 2021-08-22 by dgsiegel