showmewebcam/showmewebcam #116: UVC query errors

dgsiegel, 2021-03-22:
When booting up a Pi0 (probably happens with Pi0w and Pi4 as well), I get this in `dmesg`:

```
[  +0,007594] uvcvideo: Found UVC 1.00 device Piwebcam (1d6b:0104)
[  +0,004627] uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[  +0,007948] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
[  +0,002236] uvcvideo: Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).
[  +0,003767] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
[  +0,009293] input: Piwebcam: UVC Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1.2/1-4.1.2:2.0/input
/input57
```

Similarly, when using `v4l2-ctl -d /dev/video2 -l` you can trigger the same messages in dmesg:

```
[  +3,491996] uvcvideo: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).
[  +0,007836] uvcvideo: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).
```

These failed queries look like they come from this commit: 8ba054b25ef8a5c2993473088795deac4588c5e4

@tuyenld could you have a look into this?

tuyenld, 2021-03-22:
I'm a little busy right now, I'll take a look at weekend.

tuyenld, 2021-03-27:
I can't reproduce this message in `dmesg`. I tested with Pi Zero + v1.3 Camera on Debian.

Anyway, I think it is safe to ignore these error messages. My commit enables video controls for many type of camera as much as possible.
However, not every camera hardware supports these controls. 

For example, [Pi camera v1.3 OV5647](https://cdn.sparkfun.com/datasheets/Dev/RaspberryPi/ov5647_full.pdf) didn't support `White Balance Temperature`, but Rasp does report that it support this control , then, you got this error:
```
uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
```

dgsiegel, 2021-03-27:
It definitely happens with the HQ cam. Is there a way we could only enable the right controls for the right camera module?

wherget, 2021-03-29:
I've whipped up a small patch to allow us to set these from command line (as per earlier idea in discord), I'll have a PR ready as soon as I've verified I haven't majorly broken things.

wherget, 2021-03-30:
Given the errors you're seeing, could you test with my PR (see previous mention) and setting `usb_f_uvc.bm_controls=175,0`? (disabling bits 4,6,9 and 10; looks like the count is 0-based)

dgsiegel, 2021-04-20:
Nope, that does not seem to work:

```
[  +0,006240] uvcvideo: Found UVC 1.00 device Piwebcam (1d6b:0104)
[  +0,010522] uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[  +0,006957] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
[  +0,002523] uvcvideo: Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).
[  +0,003735] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
```
Setting `usb_f_uvc.bm_controls=0,0` doesn't seem to help either

dgsiegel, 2021-04-20:
`usb_f_uvc.proc_controls=175,0` only partially helps:

```
[  +0,008088] uvcvideo: Found UVC 1.00 device Piwebcam (1d6b:0104)
[  +0,013003] uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[  +0,005673] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
```
`usb_f_uvc.proc_controls=0,0` disables all warnings. It's probably just a matter of finding the right bit pattern.

pazz, 2021-05-25:
I am getting the same error messages. Using the HQ cam, and both stable 1.8 release as well as a39310609ff77bd71bf1044457f5928831c45c64.

It boots but eventually hangs at 

```
random: 6 urandom warning(s) missed due to ratelimiting
```

`/boot/commandline.txt` is below.

```
root=/dev/mmcblk0p2 rootwait console=tty1 console=ttyAMA0,115200 rootfstype=squashfs ro modules-load=dwc2,libcomposite usb_f_uvc.proc_controls=175,0
```

Host log:

```
[137180.837435] usb 1-4: new full-speed USB device number 48 using xhci_hcd
[137190.997438] usb 1-4: new high-speed USB device number 49 using xhci_hcd
[137191.161599] usb 1-4: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[137191.161614] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[137191.161622] usb 1-4: Product: Piwebcam
[137191.161627] usb 1-4: Manufacturer: Show-me Webcam Project
[137191.161632] usb 1-4: SerialNumber: 100000000d2386db
[137191.212159] uvcvideo: Found UVC 1.00 device Piwebcam (1d6b:0104)
[137191.240924] uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[137191.290210] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
[137191.313487] uvcvideo: Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).
[137191.335507] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
[137191.362452] input: Piwebcam: UVC Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:2.0/input/input91
[137191.363959] cdc_acm 1-4:2.2: ttyACM0: USB ACM device
```
Although it mentions the usb tty, I cannot log into it:

```
[137296.758815] cdc_acm 1-4:2.2: acm_port_activate - usb_submit_urb(ctrl irq) failed
```

dgsiegel, 2021-05-25:
> `[137296.758815] cdc_acm 1-4:2.2: acm_port_activate - usb_submit_urb(ctrl irq) failed`

this is caused by a current kernel bug: https://bugzilla.kernel.org/show_bug.cgi?id=212751

pazz, 2021-05-26:
OK so you are saying that the TTY issue is a known upstream bug, meaning that should not happen on older as well as future builds. The UVC thing is what really blocks the boot correct? 

Is there an older build that is known to work fine for the pi zero w with the HQ cam? My hardware works, I've confirmed with the full pi OS but I have not been able to get this image to produce video over UVC.

dgsiegel, 2021-05-26:
> OK so you are saying that the TTY issue is a known upstream bug, meaning that should not happen on older as well as future builds. The UVC thing is what really blocks the boot correct?

No, these are separate issues. Showmewebcam works fine in this regard, it's the kernel bug that prevents you from accessing your Pi through USB.

The query errors/warnings won't affect whether showmewebcam works or not, they just tell you that a certain UVC control is not available.

And all of those issues don't have anything to do with blocking the Pi from booting or preventing it from working as a webcam.

wherget, 2021-05-26:
I'm reading the kernel bugreport as being on the host side. (`cdc-acm` is the host, not the gadget side, the latter would be `usb_f_acm`, right?) Maybe try downgrading your host kernel? As @dgsiegel said, showmewebcam is fine there.

Also, you tried setting `usb_f_uvc.proc_controls=0,0` in `cmdline.txt`. That has changed since the last comment pertaining to that feature in this issue. You should try `usb_f_uvc.processing_unit_controls=0,0` with https://github.com/showmewebcam/showmewebcam/commit/a39310609ff77bd71bf1044457f5928831c45c64 (it also won't work with 1.8)

pazz, 2021-05-26:
Thanks for clarifying, that makes a lot of sense now. Indeed, the tty issue is with my host kernel and I can access the UVC cam without issues on a windows 10 work laptop.
Many thanks for the quick support and the awesome project!

dgsiegel, 2021-08-19:
@wherget do you think it would make sense to add `usb_f_uvc.camera_terminal_controls=0,0,0 usb_f_uvc.processing_unit_controls=0,0` as default to `cmdline.txt`? This way people wouldn't see any query errors and whoever wants to mess with it can easily change it now.

wherget, 2021-08-19:
I feel like it would be better to set the defaults in the kernel patch. I am aware we're setting some other things in `cmdline.txt` already, but the `uvc` options are kinda long-winded and have magic-seeming bitfield values where I'd prefer to not encourage people to fiddle with.

I don't want to hide the feature, it's just that I'm hoping if you explicitly want to tinker with host controls you could still find it, but there's a greater chance of having read the docs by then.

That said, even then "Nothing" seems kinda harsh (if very safe). Just checked with the older release I'm still running on one of the cams here, Brightness control should definitely work. I vaguely remember Contrast and Saturation doing something as well, but can't quickly check that. A sane default should definitely allow rudimentary host control out of the box for people not wanting to mess around with serial terminals.

Not having checked these settings, I propose `0,0,0` and `11,0` to be trialled as a middle-ground default.

dgsiegel, 2021-08-19:
> A sane default should definitely allow rudimentary host control out of the box for people not wanting to mess around with serial terminals.

Agree

> Not having checked these settings, I propose 0,0,0 and 11,0 to be trialled as a middle-ground default.

That would work. Want to propose a PR or shall I? In any case, if I understood you correctly, something like this should be enough (plus editing the Readme and the wiki):

```
diff --git a/patches/linux-custom/0001-linux-enable-more-video-controls.patch b/patches/linux-custom/0001-linux-enable-more-video-controls.patch
index 0eac227..710f626 100644
--- a/patches/linux-custom/0001-linux-enable-more-video-controls.patch
+++ b/patches/linux-custom/0001-linux-enable-more-video-controls.patch
@@ -6,14 +6,14 @@ index fb0a892..5c850be 100644
  module_param_named(trace, uvc_gadget_trace_param, uint, 0644);
  MODULE_PARM_DESC(trace, "Trace level bitmask");
  
-+unsigned int uvc_gadget_camera_terminal_controls[3] = {2, 0, 0};
++unsigned int uvc_gadget_camera_terminal_controls[3] = {0, 0, 0};
 +module_param_array_named(camera_terminal_controls, \
 +              uvc_gadget_camera_terminal_controls, \
 +              uint, NULL, 0644);
 +MODULE_PARM_DESC(camera_terminal_controls, \
 +              "Camera terminal control availabilty bitfield");
 +
-+unsigned int uvc_gadget_processing_unit_controls[2] = {255, 6};
++unsigned int uvc_gadget_processing_unit_controls[2] = {11, 0};
 +module_param_array_named(processing_unit_controls, \
 +              uvc_gadget_processing_unit_controls, \
 +              uint, NULL, 0644);

```

dgsiegel, 2021-09-01:
PR #156

ChunkLightTuna, 2022-01-14:
> `usb_f_uvc.proc_controls=175,0` only partially helps:
>
> ```
> [  +0,008088] uvcvideo: Found UVC 1.00 device Piwebcam (1d6b:0104)
> [  +0,013003] uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit [...]

 I tested each bit w/ a zero W and the hd camera.

D2(Hue) = "control 6"
D5(Gamma) = "control 9"
D6(White Balance Temperature) = "control 10". 
D9(Gain) = "control 4"

`usb_f_uvc.processing_unit_controls=155,4` is what I ended up using.