showmewebcam/showmewebcam #117: Request: Support higher resolution than 1920 x 1080 even if we have to lower frame rate

toxicated101, 2021-03-25:
I would like to use a HQ camera as a document scanner (place a piece of paper under the camera and be able to magnify the output). I know we are limited by bandwidth over USB 2.0 but most USB 2.0 cameras can do higher resolutions that 1080p by sacrificing frame rate for example 2592 (H) x 1944 (V) pixels  MJPEG 15fps  or even higher if we only want 5fps.

tuyenld, 2021-03-28:
I enabled configuration `2592 (H) x 1944 (V)` resolution at 5fps, but failed to get image output.
Here is VLC command line I tried:
```bash
vlc v4l2:// :v4l2-dev=/dev/video0 \
            :v4l2-width=2592 \
            :v4l2-height=1944 -- \
            sout="#transcode{vcodec=theo,vb=256,fps=5}"
```

wherget, 2021-03-30:
I was just casually reading `bcm2835-camera.c` (to look for the missing camera parameters) and there's `MAX_VIDEO_MODE_{WIDTH, HEIGHT}` (exposed as `bcm2835-v4l2.max_video_{width,height}` to commandline/sysfs, if I understand correctly). Might need to set that as well. Gets logged as "stills mode > WxH".

Looks to be the upper limit of when MMAL video mode is used as opposed to MMAL camera mode, but apparently only for uncompressed video? Defaults to 1280x720. How extensively did we test uncompressed video?

This [Pi forum post about MJPEG with raspivid](https://www.raspberrypi.org/forums/viewtopic.php?t=196145) might be interesting as well, money quote:

> you've told it to run at 30fps, so the sensor will only be using the 1920x1080 mode and upscaling in order to achieve that. -fps needs to be <=15 to get 8MPix off the sensor.

Does vlc actually request 5fps from v4l if the `sout` only runs on 5fps, or do you have to request `:v4l-fps=5` as well?

Also sounds like we need more `gpu_mem`?