showmewebcam/showmewebcam #123: Confusion about auto exposure mode values

taprobane99, 2021-04-04:
video quality is great during the daytime with my pi hq + 6mm lens.

If I set auto_exposure=1 in camera.txt I get slightly better quality than the default of auto_exposure=0 in the evening/dim lighting.

on the camera-ctl UI 1 is labelled "manual", but according to the UVC standard 1 is auto.

can someone clarify whether 0 or 1 is auto, and if 1 is better should this be the default?

[image]

dgsiegel, 2021-04-04:
We'd be happy to adjust the defaults if they bring better results, feel free to open a PR.

For the UVC standard you might need to open an issue for https://github.com/peterbay/uvc-gadget, as those things are set up there.

taprobane99, 2021-04-05:
On further investigation I think 0 (the default) is automatic and 1 is manual. It just differs from the UVC convention.

Will close this.

closed 2021-04-05 by taprobane99