showmewebcam/showmewebcam #125: piZeroW based webcam loads on MacOS, but not detected as USB webcam on raspberry pi 4 or 3B+

leoshmu, 2021-04-13:
Hello, 

I have a piZeroW based usbwebcam using your code, and it works when plugged into Mac OSX. However, when I plug it into a raspberry pi 4, it isn't detected when running lsusb. I thought it should be compatible to run on raspberry pi 4 running latest raspian pi OS. I ran all the updates to update the rpi and still nothing. I'd appreciate your help and can provide more info, just not sure where to get started. 
Thanks in advance for your help.

Best, 
Leo

leoshmu, 2021-04-14:
I am so sorry - it turns out I was using a micro usb cable that was just designed for power and not data, and as I was switching between raspberry pi and mac for testing, I was switching cables as well. Apologies for the silly error - it was just a poor cable!

closed 2021-04-14 by leoshmu

cloudiercom, 2021-05-28:
@leoshmu Sorry to awake this issue, did you test it with the 3B+ ? I have two of them laying around, would be awesome to use for webcam usage.

tmikaeld, 2021-05-28:
@leoshmu Sorry to awake this issue, did you test it with the 3B+ ? I have two of them laying around, would be awesome to use for webcam usage.