showmewebcam/showmewebcam #127: Multiple Cameras are misidentified by the OS

klechacz, 2021-04-15:
When multiple cameras are used in same OS it has a problem consistently distinguishing in between them. 

This means in eg. OBS if you setup 3 cameras and scenes, after reboot or eg. disconnecting USB-HUB to which they are connected the order of cameras changes. This means 'Camera2' shows as 'Camera1' in all scenes where the source is used. This is extremely inconvenient and does not seam to be happening when 'commercial' cameras are used. 

My initial though is that this is likely caused by same serial number used by all cameras while registering on USB. In upcoming days I will try to make appropriate patch and test this. If anyone have other ideas why such situation happens - let me know.

htruong, 2021-04-17:
Thanks for the cusor.

Yeah that might be it. You can read the pi serial [with something like this](
https://github.com/showmewebcam/showmewebcam/pull/73/files#diff-082bb121e6a5ae36b9adc0c15673b20506ad79b1f0b18f60ded5153b5c879b97R3) and so you'll have different serials for each camera.

It would certainly not hurt to be able to "name" the camera too, which can be sourced from a config file on `/boot`.

klechacz, 2021-04-19:
FYI I did initial build Friday by changing the serial number to use PI CPU Number in line 30 of package/piwebcam/multi-gadget.sh
```echo `cat cpuinfo | grep "Serial" | cut -d ' ' -f2`         > strings/0x409/serialnumber```
very quick initial tests are 'promissing' - but I will let you know when I have a moment to 'systematically' test it to make sure it indeed fixes the issue. 
Naming the camera or 'by default' using like 4 last characters of serial as suffix to make it 'fixed' would also make it much easier to use (eg. 'Pi Webcam A31D')  - while retaining the capability to just flush the standard image to all cameras without need of 'individual configuration provisioning'. Customization in '/boot' would allow user (if needed) to make the system label align with 'eg' labels on the cam itself.

klechacz, 2021-06-21:
Hi @htruong ,
I can confirm that adding following two lines to `/package/piwebcam/multi-gadget.sh`
```PI_SERIAL="$(grep Serial /proc/cpuinfo | sed 's/Serial\s*: 0000\(\w*\)/\1/')"
echo "5f3759df${PI_SERIAL}"   > strings/0x409/serialnumber
```
(as in your sample in the branch linked in your comment) make the behaviour much much more consistent - even though from time to time Windows still misidentifies the sources. It looks like in upcoming month I will be setting up one more unit - I will try to rename it then and report the results.

dgsiegel, 2021-08-22:
Would you mind preparing a patch for this? Sounds like a good thing to have.

klechacz, 2021-08-23:
Hi @dgsiegel,
  Below you will find diff which seams to be working for me. As to renaming the cameras I hit a small snag. Currently I have six such cameras in 'mini-studio' operated by OBS under Windows 10. Unfortunately the OS labels all cameras 'UVCWebcam' (or simmilar string as I am writing this from memory). I was not able (in time I had) to find where this string comes from.

Below diff which at least have stopped the cameras from swapping places in scenes each time I powered the hardware off and on again.  

```diff --git a/package/piwebcam/multi-gadget.sh b/package/piwebcam/multi-gadget.sh
index 2b95b13..ddeb87b 100755
--- a/package/piwebcam/multi-gadget.sh
+++ b/package/piwebcam/multi-gadget.sh
@@ -27,7 +27,10 @@ echo 0x40 > bMaxPacketSize0
 mkdir -p strings/0x409
 mkdir -p configs/c.2
 mkdir -p configs/c.2/strings/0x409
-echo `cat cpuinfo | grep "Serial" | cut -d ' ' -f2`         > strings/0x409/serialnumber
+
+PI_SERIAL="$(grep Serial /proc/cpuinfo | sed 's/Serial\s*: 0000\(\w*\)/\1/')"
+echo "5f3759df${PI_SERIAL}"   > strings/0x409/serialnumber
+#echo `cat /proc/cpuinfo | grep "Serial" | cut -d ' ' -f2`         > strings/0x409/serialnumber
 echo "Show-me Webcam Project" > strings/0x409/manufacturer
 echo "Piwebcam"               > strings/0x409/product
 echo 500                      > configs/c.2/MaxPower
```

dgsiegel, 2021-08-23:
> ```diff
+echo "5f3759df${PI_SERIAL}"   > strings/0x409/serialnumber
> ```

Where does this string come from?

klechacz, 2021-08-23:
Part comes from /proc/cpuinfo and is a derivative of pi's cpu serial - prefix comes from the code referenced by @htruong earlier in this thread. This seams to work - while there seamed to be no effect when I did use just the serial im my earlier experiments (signs of which are for some reason still seen in the diff when I look at it).

dgsiegel, 2021-08-23:
Oh right. I think in that case we could just use the full serial number, exposed by the Pi.

klechacz, 2021-08-23:
I agree. In the meantime I looked at the code again and it seams I have mistyped the path from which initial serial number (`cpuinfo` instead of `/proc/cpuingo`) was sourced - this explains why my initial experiments had not yield the expected results. 
The only possible other thing to consider (which I have originally blamed for the difference in behaviour between my initial code and one from @htruong 's branch) is if this serial in USB standard have some requirements (eg. should be some number of bytes long etc.)

dgsiegel, 2021-08-23:
Yeah there is a limit, but it is the same, so we can safely use it. I would only suggest to use `/sys/firmware/devicetree/base/serial-number` instead of parsing `cpuinfo`

klechacz, 2021-08-23:
makes perfect sense and is better candidate than `cpuinfo` which was the only one I thought of at the time. The only requirements I can thing of effectively are for this string to be:
a) different between each device when same firmeware runs
b) relatively consistent between runs and (optimally) firmware upgrades
c) it would be nice if it is not too long / too short
`/sys/firmware/devicetree/base/serial-number`  seams to satisfy all three of this requirements at a glance.

dgsiegel, 2021-08-23:
Created a PR for this: https://github.com/showmewebcam/showmewebcam/pull/155

Would you mind testing the images once they are built?