showmewebcam/showmewebcam #137: HT Avail timeout on boot

Gittinator, 2021-05-14:
The images (1.7 and 1.8) are failing to boot for me on a RPi4 and a Pi0W, both of which boot regular Raspbian fine. It is throwing the error "bcrcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50". 

Also, for the RPi4, which USB port should I be using?

Any idea what this might be caused by?

dgsiegel, 2021-05-14:
See https://github.com/showmewebcam/showmewebcam/issues/95 and https://github.com/showmewebcam/showmewebcam/wiki/FAQ#my-raspberry-pi-doesnt-work for some possible solutions.

The message you see has nothing to do with your error, it's just the last line you see in the boot up. Hence I'd guess your camera cable got loose or you might be even using a broken USB cable.

Gittinator, 2021-05-27:
That makes sense, I figured it out for the Pi0W. For the RPi4, which USB port should I be using to connect to my computer as the webcam? None of them seem to work.

Username-is-already-taken, 2021-06-18:
> The message you see has nothing to do with your error, it's just the last line you see in the boot up.

I was in the same situation, and that really threw me off. When it's your first time ever booting it, and it doesn't work, it seems reasonable to interpret that error as the cause. As it turns out, that message is on the screen even when the camera is working fine.

In my case, I had been using a USB cable that was wired for power-only. ugh, I'm going to have to start labeling those cables :/

closed 2021-11-17 by dgsiegel