showmewebcam/showmewebcam #139: "Resized root filesystem. Rebooting in 5 seconds" in a loop

foehni, 2021-05-16:
I have installed the 
dd if=sdcared-raspberrypi0w-v1.80.img of=/dev/sda bs=1M

but after inserting it into a pi zero w and booting, I get the message "Resized root filesystem. Rebooting in 5 seconds" in a loop. ist reboots and shows it again.

I have deleted the partitions with fdisk and I also formatted the whole sd card with 'mkfs.ext2 /dev/sda' and installed the image again with dd but it is still the same. 

Is there is something wrong with the image?

Regards  foehni

dgsiegel, 2021-05-16:
I really hope you didn't format your computers disk, as `sda` normally refers to the primary disk of your computer. Are you sure you used the right disk? Or are you writing it from inside a Pi?

foehni, 2021-05-16:
Hello dgsiegel, no worry - I own some more raspberry's and I am writing this on a different Raspi which is booting from /dev/mmcblk0p1 and p2!! :-D

dgsiegel, 2021-05-16:
Would you mind trying to write the image from a different computer? Would be interesting to see if the Rasperry is the problem here. It might also the be SD Card at fault, so try a different one as well.

foehni, 2021-05-16:
I have written it with a raspberry 4 via etcher and afterwards from a raspberry 3 via dd command. I have downloaded the image again from both computers. Same result.

htruong, 2021-05-16:
That does not sound like you have written the right image to the right sd card. 

I'm quite sure that we do not have anything to do with resizing rootfs. We never resize rootfs. There is nothing in our system that could have produced that message. It has to be the wrong image that you're booting from.

Please check the card you're booting from.

foehni, 2021-05-16:
Hello htruong,
I have chosen 4 different sd cards and written the image for pi zero w via etcher and dd to the sd cards and booting from it.

It is always the same. The initial information is always the 'resized ...' information and then it is booting again.

Inside the cmdline.txt in the boot directory of the sd card, you will find at the very end ...init=/usr/lib/raspi-config/init_resize.sh

Is this different from your experience?

dgsiegel, 2021-05-16:
Then it seems to me you're not writing the showmewebcam image to the SD card. The contents of 1.80 are:

`root=/dev/mmcblk0p2 rootwait console=tty1 console=ttyAMA0,115200 rootfstype=squashfs ro modules-load=dwc2,libcomposite quiet`

foehni, 2021-05-17:
Hi Daniel,
I agree with you with the cmdline.txt after the first boot. Because this initial line will be changed after the resizing is done.
Can you download the image and burn it to a SD card and before you put it into a PI zero w, mount the first partition and check the content of the cmdline.txt?

I'm 100% to write to the right device and boot from the right SD card.

btw. I have changed the cmdline.txt by removing the init... command. When I boot from it, the same things happens and the cmdline.txt looks the same as before and this init=...resize.. appears again.

I have no glue if this is getting generated during the boot process.

dgsiegel, 2021-05-17:
Yes this is from the mounted image. Also, we don't ship `/usr/lib/raspi-config/init_resize.sh` at all, so please check that you've written the right image to the sd card.

foehni, 2021-05-17:
Wow that is driving me crazy.
I was deleting the partitions with disk and it confirms it during writing to the disk. Afterwards the partition was still there!!

-------
raspi

root@raspileica:/home/pi/downloads# ls -l
total 37876
-rw-r--r-- 1 root root     4080 May 16 17:30 fdisk.txt
-rw-r--r-- 1 pi   pi   38777344 Mar 31 14:07 sdcard-raspberrypi0w-v1.80.img
root@raspileica:/home/pi/downloads# fdisk -l -u | grep sda
Disk /dev/sda: 29.5 GiB, 31657558016 bytes, 61831168 sectors
/dev/sda1         8192  532479  524288  256M  c W95 FAT32 (LBA)
/dev/sda2       532480 5623807 5091328  2.4G 83 Linux
root@raspileica:/home/pi/downloads# 

root@raspileica:/home/pi/downloads# fdisk /dev/sda

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): d
Partition number (1,2, default 2): 

Partition 2 has been deleted.

Command (m for help): d
Selected partition 1
Partition 1 has been deleted.

Command (m for help): p
Disk /dev/sda: 29.5 GiB, 31657558016 bytes, 61831168 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x6c586e13

Command (m for help): w
The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.

root@raspileica:/home/pi/downloads# 
root@raspileica:/home/pi/downloads# fdisk -l -u | grep sda
Disk /dev/sda: 29.5 GiB, 31657558016 bytes, 61831168 sectors
/dev/sda1         8192  532479  524288  256M  c W95 FAT32 (LBA)
/dev/sda2       532480 5623807 5091328  2.4G 83 Linux
root@raspileica:/home/pi/downloads# 

——————————————————————
Now I have chosen a brand-new SD card
———————————
root@raspileica:/home/pi/downloads# fdisk -l -u | grep sda
Disk /dev/sda: 14.9 GiB, 15931539456 bytes, 31116288 sectors
/dev/sda1        8192 31116287 31108096 14.9G  c W95 FAT32 (LBA)
root@raspileica:/home/pi/downloads# fdisk /dev/sda

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): p
Disk /dev/sda: 14.9 GiB, 15931539456 bytes, 31116288 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot Start      End  Sectors  Size Id Type
/dev/sda1        8192 31116287 31108096 14.9G  c W95 FAT32 (LBA)

Command (m for help): d
Selected partition 1
Partition 1 has been deleted.

Command (m for help): p
Disk /dev/sda: 14.9 GiB, 15931539456 bytes, 31116288 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Command (m for help): w
The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.

root@raspileica:/home/pi/downloads# fdisk -l -u | grep sda
root@raspileica:/home/pi/downloads# dd if=sdcard-raspberrypi0w-v1.80.img of=/dev/sda bs=1M
36+1 records in
36+1 records out
38777344 bytes (39 MB, 37 MiB) copied, 3.88285 s, 10.0 MB/s
root@raspileica:/home/pi/downloads# fdisk -l -u | grep sda
Disk /dev/sda: 14.9 GiB, 15931539456 bytes, 31116288 sectors
/dev/sda1  *        1 32768   32768  16M  c W95 FAT32 (LBA)
/dev/sda2       32769 75736   42968  21M 83 Linux
root@raspileica:/home/pi/downloads# 
 

root@raspileica:/home/pi/downloads# cd /mnt/
root@raspileica:/mnt# ls
p1  p2
root@raspileica:/mnt# mount /dev/sda1 p1
root@raspileica:/mnt# cat p1/cmdline.txt 
root=/dev/mmcblk0p2 rootwait console=tty1 console=ttyAMA0,115200 rootfstype=squashfs ro modules-load=dwc2,libcomposite quiet
root@raspileica:/mnt# 

-----------------------------
[image]

Now it is starting and showing this.

If I disconnect it and attach it to the Mac and MacBook, it is not shown as a webcam.

dgsiegel, 2021-05-17:
You don't need to format the sd card or use fdisk, dd is enough as all partition information is embedded in the image.

As for the other issue, see https://github.com/showmewebcam/showmewebcam/issues/137. Most likely your camera cable or USB cable are not attached correctly.

closed 2021-05-17 by dgsiegel

foehni, 2021-05-17:
Hello Daniel,

if I boot with a raspbian stretch light and call 'raspistill -o test.jpg', it is making a perfect photo!!

Therefore the cable should be ok and the camera. But lets check the thread #137 .

I made a comment #initial_turbo=10 in /boot/config.txt but there is no change.

I have removed the 'quiet' entry from the cmdline.txt and got the information in the picture..

[image]

dgsiegel, 2021-05-17:
Do you have a Pi0 or a Pi0w?

foehni, 2021-05-17:
Pi0w

dgsiegel, 2021-05-17:
Ok, please try a different USB cable and/or different USB port

foehni, 2021-05-17:
3 different USB cables, 3 different power supplies, tested on both USB ports on the pi0w. Same result.

Why a stretch light is starting without any issues with the same Hardware ?

htruong, 2021-05-17:
Do you mind putting the usb cable in the middle of the board, then before you plug the camera in, try running

```bash
dmesg -w
```

On your host computer. Then plug the camera in and paste the result here?

htruong, 2021-05-17:
From your screen above, I can see that the Pi0W detected a USB keyboard connected to the USB port of it.

That is the port that you use for USB-OTG. That is, you have to unplug the keyboard from that port and then connect it directly to the host computer via a micro-USB cable. Please try that with a laptop and let us know.

foehni, 2021-05-18:
[image]
[image]
[image]

I have attached different computers to the usb plug (the one more to the middle of the pi0w.

IMG_1027 was attached to a MacBook 12"
IMG_1028 was attached to a iMac
IMG_1029 was attached to a raspberry pi 3

htruong, 2021-05-18:
Please try the steps with the "dmesg -w" trick above with the raspberry pi 3 host and paste the dmesg output of the pi3.

So far your console looks normal.

foehni, 2021-05-18:
If I connect it to an raspi3 the domes -w shows 
Under-voltage detected! (0x00050005)
Voltage normalised (0x00000000)
Doesn't matter which falble I'm using.
The raspi3 is inside a pi-top with a big battery which  is connected to the pi-top power supply with an output of 18V and 3A.

If I'm connecting the zero to the Mac and call dmesg >> dimes.txt, the attached output is visible.

[dmesg.txt](https://github.com/showmewebcam/showmewebcam/files/6500078/dmesg.txt)

htruong, 2021-05-18:
I think your micro usb cable is bad and can only carry power. That's why it doesn't get defected. 

But that's just a wild guess -- I need the dmesg output of the pi 3 (not the mac).

foehni, 2021-05-18:
dmesg -w on a raspbery 4 shows no changes when I connect the pi0w via usb.

[   13.514975] random: 7 urandom warning(s) missed due to ratelimiting
[   13.863644] CUSE: failed to register chrdev region
[   14.413038] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   14.813470] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   14.813721] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   14.814112] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.867119] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   15.386538] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   19.611784] broken atomic modeset userspace detected, disabling atomic
[   20.284237] Bluetooth: Core ver 2.22
[   20.284296] NET: Registered protocol family 31
[   20.284303] Bluetooth: HCI device and connection manager initialized
[   20.284318] Bluetooth: HCI socket layer initialized
[   20.284329] Bluetooth: L2CAP socket layer initialized
[   20.284344] Bluetooth: SCO socket layer initialized
[   20.289544] Bluetooth: HCI UART driver ver 2.3
[   20.289554] Bluetooth: HCI UART protocol H4 registered
[   20.289606] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   20.289756] Bluetooth: HCI UART protocol Broadcom registered
[   20.448564] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   20.448574] Bluetooth: BNEP filters: protocol multicast
[   20.448590] Bluetooth: BNEP socket layer initialized
[   20.491923] Bluetooth: RFCOMM TTY layer initialized
[   20.491946] Bluetooth: RFCOMM socket layer initialized
[   20.491967] Bluetooth: RFCOMM ver 1.11

This is what is displayed with dmesg -w.

As well with 'tail -f /var/log/messages' nothing happens if the pi0w is connected via USB cable. 

I can see the LED on the pi0w is flashing once in the beginning and three times very quick after 15 seconds. But no information via USB.

I have tried 3 different cables.

All the cables was used with a stretch light installation on the pi0w and it is working. Raspistill and raspivid are working with all three cables. Just the boot time is around 50 seconds and your solution would much easier to handle - if it is working.

I do not know what is the difference between the pi0w and the HQ camera of mine and yours. This hardware is almost the same. The sd card is new (sandisk ultra 16GB with 98MB/s)
Did you have downloaded the image v1.80 from the git side and install it on a pi0w? I'm pretty sure there is sommething wrong with the image. I do not have an explanation why this is making so much trouble here on my side.

foehni, 2021-05-19:
Short update,

I found another 'USB cable and what should i say - it is working now.

Thank you for your support.

How can we identify which cables are working and if it is really the cable??

Even if you order a new one - I was using the newest and thickest of all the hole time - you'll never know if it is the right one.

Thanks again

htruong, 2021-05-19:
Hah... Cool to hear. Glad it worked for you. 

I usually just test all of my usb cables with something I know works (a show me webcam pi or a Arduino) then I just mark all the cables that carry power only with a label or marker. That way next time I don't accidentally use a power only cable. 

Dmesg -w usually is extremely helpful to debug usb problems as well. The rule of thumb is dmesg -w will tell you if you plug in anything at all. When dmesg -w doesn't tell you anything when you plug a usb device in, then it's likely the cable is bad or the device is dead. Since you're device wasn't dead, cable has to be bad.