showmewebcam/showmewebcam #140: Include v4l2grab in the package

Chimrod, 2021-05-20:
This ticket is not really an issue, but rather a request for a new binary in the image

[v4l2grab](https://github.com/twam/v4l2grab) is a small binary which allow to take picture for a camera on linux. It works with v4l2, which is already included in the image, and this does not increase the binary size very much.

Including the binary in the compilation is just a one line modification in the file ```configs/config```, you just have to add this line to get the binary :

    BR2_PACKAGE_V4L2GRAB=y

## What can you do?

Take a picture at full resolution : ```v4l2grab -W 3280 -H 2464 -q 100 -o /tmp/picture.jpg```, which gives better results than exporting an image from the webcam. The webcam can be used to set and adjust the scene, and then take the photo once you are ready

## But you can't export the file!

Just use the force ! (or precisely, use the serial connexion). This command will take the picture, and get the result as a serializable string:

    echo "v4l2grab -W 3280 -H 2464 -q 100 -o /tmp/picture.jpg ; base64 /tmp/picture.jpg" >> /dev/ttyXX ; cat /dev/ttyXX | tee picture.txt

You will see a lot of data flowing throught the terminal. Press ```^C``` when all the file is transfered. Now you just have to decode the result: remove the first line (the command), the last line (the shell prompt), and get the picture:

     sed '1d;$d' picture.txt | tr -d '\r' | base64 -d > picture.jpg

## And now enjoy !

of course, it would be better with a python program (or any other application) which automatize the sequence, but it's working as is.

dgsiegel, 2021-05-25:
Feel free to create a PR and maybe include a small script for the above

dgsiegel, 2021-11-17:
Closing for now, feel free to create a PR!

closed 2021-11-17 by dgsiegel