showmewebcam/showmewebcam #141: idea: support for arducam motorized focus cameras

leoshmu, 2021-05-21:
Hello,

I have come across some neat motorized focus cameras from arducam - they achieve focus with VCM, and they describe the approach here:
https://www.arducam.com/raspberry-pi-camera/autofocus/

The strategy is to send commands to the VCM across i2c, using a general call address (0x0c), and this works because the MIPI channel supports i2c on pins 13 and 14. 
`os.system("i2cset -y 0 0x0c %d %d" % (dat1,dat2))`

Is there a way to send i2c commands like this with the showmewebcam? Arducam has drop in replacement IMX219 motorized focus cameras that plug into the piV2 boards, so if you use that as the camera and connect to a piZero running showmewebcam, is there a way to get the piZero to effectively do the i2cset command so as to adjust the VCM focus?

Thanks in advance, I hope there is a way to implement this!

htruong, 2021-05-25:
`i2cset` is easy to integrate, but knowing the location of the subject and adjusting focus to the subject automatically is not trivial.

Realistically this can be achieved if someone has some time to spare, just that I am not that person due to personal engagements...

I am happy to advise someone to do this.

leoshmu, 2021-05-26:
The fact that it could be easy to integrate is exciting! The most straightforward approach is to allow one to run the i2cset command on the RPI that is acting as a uvc gadget. Is i2c enabled on the RPi that is acting as a uvc gadget, and is it straightforward to allow one to pass i2cset commands to it? Autofocus would be a separate issue and I agree is more involved.

m-bornschein, 2021-06-18:
hallo, 

es gibt schon ein projekt, was sich mit autofokus beschäftig. vielleicht könnte das für sie interessant sein?

siehe: TensorFlow Object Detection
https://github.com/EdjeElectronics

https://youtu.be/aimSGOAUI8Y

ps: ich nutzte das projekt "showmewebcam", um mein amilight mit einer web-cam zu betreiben. siehe: 
https://hyperion-project.org/forum/index.php?thread/11883-ambilight-ohne-splitter-co-einfach-mit-einer-raspberry-zero-web-cam/

-- in englisch

there is already a project that deals with autofocus. maybe that could be of interest to you?

see: TensorFlow Object Detection
https://github.com/EdjeElectronics

https://youtu.be/aimSGOAUI8Y

ps: i used the project "showmewebcam" to operate my amilight with a web cam. ;-) please refer:
https://hyperion-project.org/forum/index.php?thread/11883-ambilight-ohne-splitter-co-einfach-mit-einer-raspberry-zero-web-cam/

jottr, 2021-08-17:
Germans who can't read English can use some web translator of their choice in case they don't understand something.  Please don't pollute issue trackers with your personal language preference on international projects like this.  
Greetings from Germany. :)

geko4, 2021-12-12:
Hi,

I created a PTZ (Pan-Tilt-Zoom-Focus) camera based on this project. You can take a look here:

https://www.thingiverse.com/thing:5162674

I used one of the cheap motorized lenses (6-22mm). The stepper motors are 2-phase 4-wire motors which can be driven by a double h-bridge which can be attached to 4 GPIOs of the raspberry each (one for focus and one for zoom)

the python scripts are also on thingiverse. 

The biggest issue with these lenses is that they does not hold the focus if you zoom in/out.

Greetings from Bavaria!

geerlingguy, 2022-01-03:
ArduCam also recently announced the [16 MP High Resolution Autofocus Camera for Raspberry Pi](https://www.arducam.com/16mp-autofocus-camera-for-raspberry-pi/), which would be amazing as an autofocusing webcam option. I've been playing around with their software, and basically you can fully control focus in software (but it's complicated), or use their fork of libcamera and make it autofocus at set intervals (though when it does an autofocus, it hunts back and forth quickly for a second or two, not sure what kind of algorithm it uses).

It only seems to focus on whatever's in the center of the frame (and a rather small focus area at that), but it does focus pretty quickly. Even if the camera autofocused once at startup, that could be better than no focusing at all.

ck3mp3r, 2022-02-10:
yeah, @geerlingguy after backing the project I have two of those cameras now and would have loved to use them with my iSight raspberry pi zero conversion...

diegorivasn, 2022-02-12:
> yeah, @geerlingguy after backing the project I have two of those cameras now and would have loved to use them with my iSight raspberry pi zero conversion...

Same here! What would be the steps to make this camera ([16 MP High-Resolution Autofocus Camera for Raspberry Pi](https://www.arducam.com/16mp-autofocus-camera-for-raspberry-pi/)) compatible with showmewebcam? Happy to give it a go if anyone gives me some direction on where to start

leoshmu, 2022-02-12:
In arducam blogs they show how you can send commands across i2c to control
the focus mechanism. Not sure if that same interface is used in these new
cameras (i backed it as well and hope to get them in the mail soon!), but
finding a way to send commands to the camera  from the usb camera would be
great