showmewebcam/showmewebcam #145: Unable to login into serial console

josefe17, 2021-06-10:
I was able to install the camera and use it, but I am unable to login into the serial console to adjust  the settings.

I can open the serial port at 112800 bps as described, but then no prompt appears. I try to log in as described  by typing "login root", then it prompts for password but after typing "root" as described in the Windows 10 instructions and hitting enter nothing happens until a minute later when login timeouts and the entered password is parsed as another (unknown) command.

dgsiegel, 2021-06-10:
What version are you using? With 1.80 there is no login necessary. Also, use 115200 bps instead of 112800.

htruong, 2021-06-13:
I will assume that this issue is due to the typo. If we don't hear back within a wrek from now, let's close this issue.

josefe17, 2021-06-14:
Fixed. I followed this documentation: https://tutorial.cytron.io/2020/12/29/raspberry-pi-zero-usb-webcam/ Maybe it should be updated with the better badurate and that since v1.8 login is not required.

closed 2021-06-14 by dgsiegel